UltimateHackingKeyboard/firmware #703: Missing triangle icon when recording macros

tunney, 2023-09-26:
Until the latest firmware (10.2.0) when one records a runtime macro the triangle icon lit up. Now this icon does not light up and the phrase "REC" shows on the LED display. When there are warnings in ones macros this means that the display alternates between REC, WRN, QWM — so it can be hard to determine when one is actually recording a macro.
Can the triangle be brought by for this use case?

kareltucek, 2023-09-26:
Well, the adaptive mode led was originally stolen for that purpose, so I guess this is up to @mondalaci ...

What I can do is to shorten the flashing period, which I was thinkingnabout doing anyways 

Also, when there are warnings or errors, you are supposed to resolve them, not just ignore!

kareltucek, 2023-09-26:
Or maybe we should simply make REC override the keymap abbreviation?

tunney, 2023-09-26:
I think have REC override would help here.

As for resolving warnings :)
Eagerly awaiting a new firmware to get rid of the annoying WRN :)

[image]

kareltucek, 2023-09-26:
Ah, I see. Sorry about that!

mondalaci, 2023-09-26:
Given that adaptive mode will be implemented eventually, I'd prefer using the triangle icon for this purpose as originally intended instead of repurposing it. I prefer the REC override solution. However, if users find repurposing the triangle icon for recordings useful, I'm not against it, but it should be explicitly set via smart macros.

I'm unsure whether it's planned, but we should disallow users to silence warnings because deprecation warnings can become deprecation errors in the future, so it's better to urge users to fix their commands.

mhantsch, 2023-09-27:
Here's my suggestion: blinking triangle = recording; steady triangle = adaptive mode

mondalaci, 2023-09-27:
Works for me, and it can behave the way suggested by default.

mhantsch, 2023-09-27:
The blinking/steady solution for the triangle also helps keeping the text display uncluttered. Otherwise it starts to get overloaded with too much information. I think it's enough to show the keymap name, and ERR/WRN overload as the only exception.

PR #721 (2023-10-26): Flash adaptive mode led when recording runtime macros.
Closes #703.
closed 2023-10-26 by mondalaci