UltimateHackingKeyboard/firmware #705: Mac Fn key

mondalaci, 2023-10-01:
We should try to emulate the Fn key of Mac keyboards.

https://apple.stackexchange.com/questions/340607/what-is-the-hex-id-for-fn-key might help.

simisoft-exo, 2023-10-23:
Does this mean that Mac users would get the Mac Fn key together with the UHK Fn layer?

mondalaci, 2023-10-23:
The two keys will be separately assignable if we figure this out.

yyolk, 2024-02-15:
This is also shared with the "Globe" key <kbd>🌐</kbd>, which AFAICT started showing up on the Magic Keyboard. Looks like starting in macOS Sonoma is referred to as the "Globe" key and is configurable to pop up the emoji picker, and some other options with a single keypress, when not used as a modifier for other keys.

(<kbd>⌘</kbd> + <kbd>⌃</kbd> + <kbd>␣</kbd> will still open it as well)

[image]
[image]

This article talks about some of the options it's configurable for beyond a single keypress by itself, not as a modifier

https://tidbits.com/2023/11/16/the-hidden-secrets-of-the-fn-key/

geggo98, 2024-03-12:
I think the Apple "globe button" is just "AC Next Keyboard
Layout Select" (`0x029D`) from the ["Generic GUI Application Controls"](https://usb.org/sites/default/files/hut1_2.pdf#page=153). But I don't know if that also covers the "fn" function of that button or if it is reported differently.

rdpate, 2025-05-10:
> See also the [official Apple documentation](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf#page=86).

It appears that PDF changed, and the page is now 96-97 instead of 86 in the above link -- see table 14-3 "HID Consumer Page (0x0C) controls for use by keyboards".