UltimateHackingKeyboard/firmware #706: Logical operators && and || missing

tunney, 2023-10-02:
The release notes for 10.2.0 mention and && and || but I cannot see them in the syntax for 10.2.0 or 10.3.0
Their presence in a macro does not cause an issue but they don't seem to work. At least the && tends to operate as an "or".

[image]

[image]

Will ands and ors in the if expressions be supported? Or are the release notes ambitious?

kareltucek, 2023-10-02:
Ouch. Thanks for report!

kareltucek, 2023-10-02:
Release notes are fine, it is the release itself that is not fine, as well as its documentation 😅.

PR #707 (2023-10-02): Fix && macro command.
Typo fix...

- `write "$(1 && 0)"`
- observe 1 being produced whereas 0 is correct

Closes #706