UltimateHackingKeyboard/firmware #711: Latest Firmware exception with `onInit` Macro

mohitt, 2023-10-16:
With the firware version `10.3.0` I started seeing following error in my `onInit`

```
Error at $onInit 1/3/22: Unrecognized command: ScrollSpeedDivisor
```

My `onInit` Commands are

```
set module.trackball.baseSpeed 1.1
set module.trackball.speed 2.3
set module.trackball.ScrollSpeedDivisor 16
```

Documentation says it is supported

Before submitting a new issue, make sure to do the following:

1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
3. Our contact form will suggest you solutions, so copy-paste your message at https://ultimatehackingkeyboard.com/contact-us - Please don't post customer support issues here.

kareltucek, 2023-10-16:
> Documentation says it is supported

No, it does not!

The syntax is case sensitive. Should be:

```
set module.trackball.scrollSpeedDivisor 16
```

mohitt, 2023-10-16:
[image]
This is what I see in documentation and surprisingly it was working in version 9

closed 2023-10-16 by mohitt

kareltucek, 2023-10-16:
Ahh, this documentation. Will fix that.