UltimateHackingKeyboard/firmware #713: Falsey warning – Unprocessed input encountered.

Galathius, 2023-10-18:
After the recent firmware upgrade (probably from 10.1.0 version to 10.3.0) I start getting this error for quite simple macros, which still works:

```
Warning at SmartMouseLayer 1/2/32: Unprocessed input encountered.
> 2 | ifNotInterrupted unToggleLayer mouse
>   |                                ^
Warning at SmartMouseLayer 1/3/26: Unprocessed input encountered.
> 3 | ifReleased unToggleLayer mouse
>   |                          ^
```

Macros itself:
```
holdLayer mouse
ifNotInterrupted unToggleLayer mouse
ifReleased unToggleLayer mouse
ifDoubletap toggleLayer mouse
```

The error points to line 2 which was literally copied from [this doc](https://github.com/UltimateHackingKeyboard/firmware/blob/683b044563ec578c2c7b6dc439e0ce4e40982333/doc-dev/user-guide.md?plain=1#L284).

After calling the macro I get this, but it still works as expected:
```
Error at SmartMouseLayer 1/2/32: Unrecognized command: mouse
> 2 | ifNotInterrupted unToggleLayer mouse
>   |                                ^
Error at SmartMouseLayer 1/3/26: Unrecognized command: mouse
> 3 | ifReleased unToggleLayer mouse
```

Were there any recent changes related to those commands? Or it's some kind of bug?

kareltucek, 2023-10-18:
There was just a change to the macro parser, which is now strict about unrecognized stuff left at the end of line.

unToggleLayer does not take any arguments, so `mouse` is redundant, and the engine complains about that.

Galathius, 2023-10-18:
@kareltucek thanks for such a quick answer! Removed `mouse` from line 2 and 3 and seems it works perfectly now!

closed 2023-10-18 by Galathius