UltimateHackingKeyboard/firmware #729: MacOS Sonoma Keyboard shut-down on lock

emin-grbo, 2023-11-30:
Sometimes (1-3 times a day), after my computer was in locked state, I need to plug out/in the keyboard in order for it to work.
I have a V2 keyboard, bought less than 6 months ago I think.

System: macOS Sonoma (it actually started after the latest update)
Agent version: 3.2.1

mondalaci, 2023-12-01:
Does the locked state put USB devices to sleep? I'm asking because if not, this detail is irrelevant, and your UHK would reproduce the issue regardless when not used for a while.

mondalaci, 2023-12-22:
Closing because of the lack of replies.

Feel free to follow up regardless.

closed 2023-12-22 by mondalaci

emin-grbo, 2023-12-23:
Sorry for the delay, just saw the closure notification now 😅
I am not sure as the only plugged device is the keyboard. I have a bluetooth trackPad next to it and that's it.

However, not sure it happened after the latest OS update, so I will try to track occurrences with more details and see if its reproducible more consintently.