UltimateHackingKeyboard/firmware #730: Firmware v10.4.0/Mac OS Sonoma regression: zoom key combo

robbievogt, 2023-11-30:
I have had the following stanza in my `$onInit` for probably 6 months:

```
set module.keycluster.scrollSpeedDivisor 10
set module.keycluster.navigationMode.base zoomMac
set module.keycluster.navigationMode.mod zoomMac
set navigationModeAction.zoomMac.up none
set navigationModeAction.zoomMac.down none
set navigationModeAction.zoomMac.left keystroke AG-equalAndPlus
set navigationModeAction.zoomMac.right keystroke AG-minusAndUnderscore
```

It uses the keycluster trackball left/right for MacOS screen magnifier zoom in/out.

This has worked very well until I updated firmware today 10.1.0 --> 10.4.0.

Now it seems to be regularly dropping the modifiers, resulting in typing "-" and "=" instead of zooming in/out. 

This is an extremely annoying regression, as this is functionality that I use constantly (i.e. often multiple times per minute)!

I'm on MacOS Sonoma 14.1.1.

robbievogt, 2023-11-30:
Partial workaround: I've updated to trigger macros instead with:

```
set navigationModeAction.zoomMac.left macro ZoomIn
set navigationModeAction.zoomMac.right macro ZoomOut
```

With the `ZoomIn|Out` macros just defined as tap key events with appropriate combo.

This avoids the issue with typing extraneous characters, however, it's a lot less responsive and seems to be rate-limited.

Seems like this is an issue with delays or rates of events?

kareltucek, 2023-12-01:
Interesting...

I can reproduce it. Seems like a nonzero keystrokeDelay induces it.

Do you have any other settings defined in your $onInit?

robbievogt, 2023-12-03:
Yes, I tried a couple of different `keystrokeDelay` values (10, 20) and this seemed to be exacerbating the issue so I removed it again. Currently no `keystrokeDelay` set.

Full `$onInit` (with macro workaround):

```
set leds.fadeTimeout 5

// set keystrokeDelay 20

set module.keycluster.scrollSpeedDivisor 10
set module.keycluster.navigationMode.base zoomMac
set module.keycluster.navigationMode.mod zoomMac
set navigationModeAction.zoomMac.up none
set navigationModeAction.zoomMac.down none
// set navigationModeAction.zoomMac.left keystroke AG-equalAndPlus
// set navigationModeAction.zoomMac.right keystroke AG-minusAndUnderscore
set navigationModeAction.zoomMac.left macro ZoomIn
set navigationModeAction.zoomMac.right macro ZoomOut

set module.trackball.baseSpeed 0.7
set module.trackball.speed 0.5
set module.trackball.xceleration 1.4

set secondaryRole.defaultStrategy advanced
set secondaryRole.advanced.timeout 300
set secondaryRole.advanced.timeoutAction secondary
set secondaryRole.advanced.triggerByRelease 1
set secondaryRole.advanced.safetyMargin 20
```

Also, I've just now tried explicitly with `set keystrokeDelay 0` and setting back to non-macro shortcut, and this seems to be working perfectly. Is there some default `keystrokeDelay` if it's not set explicitly?

kareltucek, 2023-12-03:
Default is 0, but UHK does not reset its config on config saves (only on power cycles)... (Yes, this is non-intuitive and should be solved eventually.)

kareltucek, 2024-01-11:
Well, this was introduced in [10.4.0](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v10.4.0), as part of _Clear sticky modifiers when navigation mode is active._. 

Anyway, it should be fixed in the next release. 

Sorry for the long wait and thanks for report!

closed 2024-01-11 by mondalaci