UltimateHackingKeyboard/firmware #731: Firmware 10.4.0 regression: no longer parses `@0`

robbievogt, 2023-11-30:
I have several macros implementing "sticky" layer modifiers, that rely on the macro time machine and wait loops.

E.g. 
```
postponeKeys ifNotPending 1 ifNotPlaytime 2000 goTo @0  // wait up to 2 sec for a single pending key
```

These stopped working with firmware update to 10.4.0. The error console in UHK Agent (nice addition!!!) complains that it can't parse `@0`. 

There are workarounds for this (e.g. labels), but it left my keyboard in an unusable state after the firmware update due to the regression.

kareltucek, 2023-12-01:
Yes, please replace it by `$currentAddress`. 

(Closing, but feel free to follow up!)

closed 2023-12-01 by kareltucek

robbievogt, 2023-12-03:
Thanks Karel, makes sense.

Not there are a couple of references to the `@` syntax in the docs still:

* user-guide.md, line 717
* reference-manual.md, line 255

kareltucek, 2023-12-03:
Thanks! 

The first one is fixed.

The second one is in "removed" command section, therefore intentional ;-).