UltimateHackingKeyboard/firmware #733: Add module and some other settings to user configuration

mondalaci, 2023-12-12:
Instead of `moduleConfigurationCount = ReadCompactLength(buffer);` let's use a `uint8_t` because we'll never exhaust it. Its value has always been zero so far.

Please rewrite `parseModuleConfiguration()` according to the following. (It has never been used so far because Agent serialized an empty module array, so backward compatibility is not an issue.)
- Leave `uint8_t id = ReadUInt8(buffer);` module id as it is.
- Next up, read the 8 LAYERID_BASIC navigation mode values.
- Next up, read `uint8_t modulePropertyCount`.
- Next up, read the module properties, which are key-value pairs. The key is always a `uint8_t`. The length and type of the value is dependent on the key.
  - Make module-independent property keys start from 0 and increment, such as baseSpeed=0, speed=1, ...
  - Make module-dependent property keys start from 255 and decrement, such as pinchZoomDivisor=255, pinchZoomMode=254, ... - Module-dependent properties of different modules can overlap to make the best use of the key id space.

Add the following to the user configuration to positions where they're consistent with already existing properties:
- secondaryRole.*
- mouseKeys.{move|scroll}.axisSkew
- diagonalSpeedCompensation
- doubletapTimeout - Is it the same as doubleTapSwitchLayerTimeout? If so, don't re-add it.
- keystrokeDelay

kareltucek, 2023-12-23:
What will be the new config version?

mondalaci, 2023-12-23:
This being a breaking change, the new user config version is 7.0.0

PR #737 (2024-02-26): Add new 7.0.0 data model config values.
Closes #733.

(Untested.)
closed 2024-02-26 by mondalaci