UltimateHackingKeyboard/firmware #734: Fine-tuning trackpoint auto drift recovery

mondalaci, 2023-12-14:
The trackpoint auto drift recovery implemented in firmware 10.5.0 works very well in my experience, but it usually takes about 5 seconds to kick in. Which value(s) should I increment/decrement to reduce this delay?

Also, what could be the unwanted consequence of shorter delays? False positive recoveries, leading to unintended trackpoint resets, which can make the trackpoint skip a beat?

kareltucek, 2023-12-14:
You can try lowering `DRIFT_RESET_PERIOD`. 

The unwanted consequence is getting drifts more often.

From my testing, the current increased tolerance of the algorithm already causes drifts to appear significantly more often than before.

mondalaci, 2023-12-21:
I've been using DRIFT_RESET_PERIOD=1000 for a week. Drift recovery happened noticeably quicker, and I haven't noticed more drifts as a side effect.

 Moving on to DRIFT_RESET_PERIOD=500, and I'll share my observations.

mondalaci, 2023-12-28:
500 seems to produce more drifts, so 1000-2000 is probably the sweet spot.

Are you sure that 1000 produces more drifts for you than 2000?

kareltucek, 2023-12-28:
I didn't test 1000 vs 2000 difference, and am probably biased in this anyways, so no I am not sure.

mondalaci, 2023-12-28:
Would you please test 1000? If you also agree that 1000 doesn't introduce more drifts, then it'd be a better choice due to its quicker recovery.

kareltucek, 2023-12-28:
Sure.

kareltucek, 2023-12-29:
Well, I didn't encounter significantly more accidental drifts than usual, although undoubtedly triggering drifts is now  much easier than before (for instance when done intentionally).

My guess is that there will be some portion of users - maybe 1 in 10 - who will spot an increase because of the way they use the trackpoint...

But as far as my opinion goes, feel free to release with 1000.

mondalaci, 2023-12-29:
Just set it to 1000. I think it's a slight improvement, and users will tell us if not.

closed 2023-12-29 by mondalaci