UltimateHackingKeyboard/firmware #736: Right click menus on macOs can often disable the keyboard

tunney, 2023-12-18:
I have only started to notice this recently.
V2 connected directly to macbook pro. Latest versions of OS and the like.

Sometimes when I right click on items, happens in all manner of applications, I get the keyboard no longer responding at all and I have to unplug it and plug it back in.
Not every time I right click on something but very very frequent

tunney, 2023-12-18:
Latest firmware as well

tunney, 2023-12-18:
On closer inspection I fear that this is a hardware issue. Apologies for raising in error.

mondalaci, 2023-12-18:
I also believe that this is a hardware issue. Feel free to reach out to [our support](https://ultimatehackingkeyboard.com/contact-us).

closed 2023-12-18 by mondalaci