UltimateHackingKeyboard/firmware #755: setVar error

Festiu, 2024-02-14:
I'm not sure if this is an unusual error, or if I simply don't understand how variable handling works.

I'm attempting to declare a variable using `setVar var_name value`  but I'm encountering an error: Too many variables. Can't allocate

I've recorded a video:
https://streamable.com/rikmeu

I'm on Agent 3.3.0 with firmware 10.5.0

kareltucek, 2024-02-14:
Well, there are 32 slots reserved for variables. You cannot create more than that.

mondalaci, 2024-02-14:
@kareltucek How about expanding the error message to: Can't allocate more than 32 variables. - It'd be more explicit.

kareltucek, 2024-02-14:
@Festiu I just can't reproduce that. Please, attach the problematic UserConfig.json, maybe something pops up there.

Also, after exporting it, please, try importing it back again in your Agent and retry.

Festiu, 2024-02-14:
I attach my .json configuration but it is not problematic but on the contrary, because once imported, the setVar on it seems to work perfectly. The problem is that I can't edit the variable names, nor add new ones, nor delete them, because then the errors appear.

[UserConfiguration 2024-02-13 (Agent 3.3.0 Firmware 10.5.0).json](https://github.com/UltimateHackingKeyboard/firmware/files/14289241/UserConfiguration.2024-02-13.Agent.3.3.0.Firmware.10.5.0.json)

The .json is a little bit long, but it doesn't do anything strange. It does have a lot of // comments, but I don't think that's a problem either, right?

I know it may sound silly, but I have suspicions that those macros named !TEMP may have caused some bug, because for example if I modify or delete them, that's when the errors appear.

See the following video:
https://streamable.com/4b8tng

But even if the !TEMP macros are the cause. Anyway, it seems that my keyboard has been bugged, because as I showed in the first video, the setVar gives error even with the configuration has been reset.

I've tested with Agent 3.2.0 too, but still the same.

This situation it's so strange.... sorry.

closed 2024-02-14 by Festiu

kareltucek, 2024-02-15:
Strange. Please let us know if it reappears.