UltimateHackingKeyboard/firmware #759: UHK v1 lost baklight on "H" key after 11.0.0 upgrade

rnurgaliyev, 2024-03-09:
I have UHKv1 with backlight mod. After updating firmware to 11.0.0, LED under "H" key stopped working. Rolled back to 10.6.0 - works correctly again. Maybe it is just me issue, since it is somewhat custom UHKv1 which was given to me by @mondalaci to help with schematics troubleshooting.

kareltucek, 2024-03-10:
So your UHK has some prototype PCB and was hand-assembled before start of the main crowd-supply production batch?

The difference between 10 and 11 firmwares is that 10 turns on all led slots (column/row adresses), while 11 turns on only those led slots that are supposed to be occupied.

So unusually wired PCB is a very likely cause.

kareltucek, 2024-03-10:
If the hypothesis is correct, then you can try compiling custom firmware with different addresses at https://github.com/UltimateHackingKeyboard/firmware/blob/7f4ec342d74d4b8773a328bca9e42db8d199b20e/right/src/ledmap.c#L68 .

kareltucek, 2024-03-10:
Edit: actually, my v1 H also does not light up. Never noticed it. Thanks for report, I will look into it!

kareltucek, 2024-03-10:
Yep, indeed wrong address. Will be fixed in the next release!

closed 2024-03-10 by mondalaci

mondalaci, 2024-03-10:
Just released firmware 11.0.1 which contains the fix.

PR #761 (2024-03-10): Fix H key backlight on V1 keyboards.
Closes #759