UltimateHackingKeyboard/firmware #760: Software KVM not detecting Trackball

mohitt, 2024-03-10:
Hi, 

I am using - https://github.com/htrefil/rkvm to switch between my two machines. Now this software KVM detects my USB mouse and mouse works as expected. But, neither my keyboard's mouse layer nor the trackball works on the client machine for rkvm.

I have installed the latest firmware and latest agent. Can someone please help me to debug whats going on. Why my regular USB mouse works where as neither the trackball module nor the mouse layer of keyboard works.

Before submitting a new issue, make sure to do the following:

1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
3. Our contact form will suggest you solutions, so copy-paste your message at https://ultimatehackingkeyboard.com/contact-us - Please don't post customer support issues here.

kareltucek, 2024-03-10:
Please, see https://github.com/UltimateHackingKeyboard/firmware/issues/536#issuecomment-1971088193 and try the firmware that is linked there.

Closing since it is basically a duplicate. Feel free to follow up though.

kareltucek, 2024-03-11:
Allright, this is the  a question for @mondalaci or Benedek.

mondalaci, 2024-03-11:
N-key rollover usually causes the issue. But if rkvm doesn't even work with the 6KRO UHK firmware version, its USB HID report parser must be very limited, and it'd also have problems with many advanced input peripherals.

I should also mention that even NKRO wouldn't be a problem if KVMs' USB HID report parsers were more capable.

@mohitt Please contact the developers of rkvm to make their report parser more capable.

I'm closing this issue because solving such issues would involve endlessly experimenting with the limitations of various KVM switches. Feel free to follow up regardless.

closed 2024-03-11 by mondalaci