UltimateHackingKeyboard/firmware #762: First keypress ignored after keyboard inactive

rightaditya, 2024-03-11:
On my UHKv1, I've been finding that the first keypress after a very brief lack of keyboard activity (can't tell exactly what the time frame is, but I'm fairly certain that it's less than a minute) gets dropped or otherwise ignored. This is a new issue, with the only thing I can think of that changed recently is that I upgraded the firmware to v11.0.0. (I've just updated to 11.0.1 and replicated it there as well.)

From what I can tell, the amount of inactive time isn't consistent. Sometimes it's around 10-20s, but I just did a quick 30s test and that didn't seem to trigger the issue.

My keyboard is connected via a CalDigit TS4 (Thunderbolt) dock. The problem doesn't appear if the keyboard is connected directly to the computer.

mondalaci, 2024-03-12:
Does it also happen with [this firmware build](https://github.com/UltimateHackingKeyboard/firmware/issues/536#issuecomment-1971088193)?

rightaditya, 2024-03-13:
I've just flashed it and will report back. This issue is more intermittent than it first seemed—I only noticed it once in the last 48 hours.

For reference, when I first started encountering this issue (shortly after the 11.0.0 release), I was able to replicate it on a different machine running Windows but through the same dock (though the machine running Windows doesn't have Thunderbolt, the dock's USB seemed to work just fine). On this machine, too, plugging the keyboard in directly to the machine solved the problem. I used the keyboard without the dock (on both machines) for a week or so without issue before plugging it back into the dock a couple days ago.

rightaditya, 2024-03-20:
Looks like it's happening again on that firmware build.

As a sanity check, I'm going to downgrade to 10.5.0 and make sure it doesn't show up there. I'll report back.

rightaditya, 2024-03-20:
...looks like I *hadn't* imagined a message around when I upgraded to 11.0.0 about being unable to revert. Is that right?

mondalaci, 2024-03-20:
You should be able to revert. If you revert, 10.6.0 is worth trying.

In any case, the root cause of the issue is your dock, which doesn't handle USB communication correctly. Please raise an issue with them.

rightaditya, 2024-03-20:
Will do. It failed earlier and put my keyboard in a state Agent didn't like, but I'm guessing that might've been this issue presenting itself during the update 😱 (Yeah, I need to remember to plug the keyboard directly into the machine when updating...)

FWIW, I flashed 11.0.2 and the problem has been very obviously noticeable since. It's either much worse than the modified firmware you linked, or else it's some really unlucky timing that makes it seem as such. I'm starting to wonder if I might've been imagining the issue on the modified firmware...

I'll try again with 10.6.0 as well as the modified firmware and see if I can pin down which (if any) the issue turns up in.

kareltucek, 2024-03-20:
> Will do. It failed earlier and put my keyboard in a state Agent didn't like, but I'm guessing that might've been this issue presenting itself during the update 😱 (Yeah, I need to remember to plug the keyboard directly into the machine when updating...)

Well, that is expected during downgrades. I think following procedure should work:

- backup your configuration
- flash a manually downloaded older firmware using up-to-date agent (https://github.com/UltimateHackingKeyboard/firmware/releases)
- (as downgraded right half may uses older communication protocol, it may have stopped communicating with the up-to-date firmware, so downgrade of module firmwares may have failed)
- (firmware now complains about incompatible config format and agent either does not work or wants to update firmware back to higher firmware)
- disable agent auto-update
- download older agent that corresponds to the older firmware (https://github.com/UltimateHackingKeyboard/agent/releases)
- (now agent complains about incompatible config format too)
- reset config using agent
- in case modules don't work, reflash the old firmware using the old agent (in case that during original flashing only right half succeeded)

If flashing does not work for some reason, the unbricking procedure using the corresponding agent version usually helps (https://ultimatehackingkeyboard.com/repair/unbricking-the-right-keyboard-half)

rightaditya, 2024-03-21:
@kareltucek Wow, thanks for the detailed instructions! Flashing with 11.0.2 was immediately noticeably worse than the modified build, and it's continuing to drive me mad, so I'm going to go and flash 10.6.0 now. I'll give that a solid test and then go back to the modified firmware for a final comparison before I report back.

rightaditya, 2024-03-21:
FYI, downgrading wasn't too crazy this time. I made sure to plug the keyboard directly in to my machine. So either the issue kicked in on the dock when I tried this before, or there was a bigger issue caused by 10.5.0 (that I used before) that isn't there with 10.6.0. I did have to use the older version of Agent, and it complained about the configuration, but it offered to restore from 2023-12-25 (before the upgrade, I guess) so it all seems good for now.

mondalaci, 2024-04-22:
Closing this issue, but feel free to follow up.

closed 2024-04-22 by mondalaci