UltimateHackingKeyboard/firmware #765: Recent firmware broke mouse keys control when using it with bluetooth dongle

yihou, 2024-03-27:
Before submitting a new issue, make sure to do the following:

1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
3. Our contact form will suggest you solutions, so copy-paste your message at https://ultimatehackingkeyboard.com/contact-us - Please don't post customer support issues here.

I have been using a Bluetooth device from Handheld Scientific that was highly recommended from the start. 

Previously, the mouse keys were functioning properly when I connected the keyboard through a Bluetooth dongle. 

However, after updating the firmware to version 11.0.0, the mouse keys no longer respond to left and right movements, only allowing the mouse cursor to move up and down.

Do let me know any details that I can provide to assist you guys to debug the problem.

yihou, 2024-03-27:
Updates: 

Just realised there's 11.0.2 firmware, installed and the issue persist.

I noticed that when I use a keyboard connected through a Bluetooth dongle, pressing the left and right buttons on the mouse causes the cursor to move up and down. Conversely, pressing the up and down buttons on the mouse results in scrolling. 

This issue also impacts trackball movements, causing it to move up and down when I scroll the ball left and right.
 
Here is my Mouse keymap settings: 
[image]

mondalaci, 2024-03-27:
This issue is indirectly caused by the following modification mentioned in the changelog of the UHK firmware 11.0.0:

> Reorder USB interfaces to achieve better compatibility with KVM switches and docking stations.

However, the updated USB interfaces of the UHK are still legit. Here's the [relevant commit](https://github.com/UltimateHackingKeyboard/firmware/commit/8f25f54d6a5666cffe87760a285338ca27c8a312).

The direct cause is that the BT-500 probably implements a non-general, device-specific USB parser that relies on the specific USB interface order of pre-11.0.0 UHK firmware versions.

We won't address the shortcomings of third-party devices. Please inform Handheld Scientific about this issue so they can update their firmware.

closed 2024-03-27 by mondalaci

yihou, 2024-03-27:
Alright, will approach Handheld Scientific then.

asdf4w3t5, 2024-04-26:
@yihou Was this mentioned to handheld scientific? 
Do they have a solution?
It's unfortunate the UHK site recommends this device but now it doesn't work lmao

ADevAJ, 2024-08-17:
Would like an update about this too

yigitkonur, 2024-11-27:
Will appreciate a fix or a fork that uses BT-500 compatible USB parser configs - as downgrade is not possible now, I guess it is more needed. @mondalaci

mondalaci, 2024-11-28:
A downgrade should be possible, according to https://github.com/UltimateHackingKeyboard/agent/issues/2361#issuecomment-2505673352

We shouldn't work around the shortcomings of third-party products, such as the USB HID parser of the BT-500. They should update their firmware if they want to support the UHK mouse USB HID interface with recent UHK firmware versions.