UltimateHackingKeyboard/firmware #770: UHK still recognized as a game controller with firmware 11.0.2

matthjes, 2024-04-20:
Hi,

I've already talked with László Monda about this:

My UHK is still recognized as a game controller in Windows 10 with firmware 11.0.2:

[image]

Steam also shows the UHK three times as a controller:

[image]

Here is a screenshot showing the currently flashed version:

[image]

I don't know what further information you require, please just ask, I'll provide it.

mondalaci, 2024-04-20:
@benedekkupper How can this happen since https://github.com/UltimateHackingKeyboard/firmware/commit/45fd12a188548d0a55cfe5cd4ade59261f1f4dc0?

benedekkupper, 2024-04-21:
I'm as clueless as you are.

@matthjes please follow these instructions to [install usbview](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview). When you run this program, you can inspect the USB bus tree of your computer, and it allows you to inspect the descriptors of your USB devices. Please send us a screenshot of the UHK device's view.

benedekkupper, 2024-04-22:
Well according to this the UHK isn't presenting a gamepad interface to your computer. Can you test it on another Windows machine, preferably with which you didn't use it before? I can imagine that a stray registry entry is the root cause of this.

matthjes, 2024-04-22:
This registry tip was a good one. I've searched for "UHK" the registry and there were two entries with "Controller (UHK 60 v1)" in their title (like in the screenshots above). I've deleted the entries and now it no longer shows up in Windows or Steam.

Thanks!

closed 2024-04-22 by matthjes