UltimateHackingKeyboard/firmware #772: Unexpected mouse button behavior (mouse buttons 4 - 20)

nathanph, 2024-04-23:
Binding a key to `MOUSE_BUTTON_4` (which I believe is typically understood to be `MOUSE_WHEEL_UP`) emits `button 8`.
This is true for all `mouseBtn#` values.

Is this expected behavior?

**`xev` output**

```
ButtonPress event, serial 37, synthetic NO, window 0x4000001,
    root 0x275, subw 0x0, time 38927967, (108,108), root:(278,259),
    state 0x0, button 8, same_screen YES

ButtonRelease event, serial 37, synthetic NO, window 0x4000001,
    root 0x275, subw 0x0, time 38928080, (108,108), root:(278,259),
    state 0x0, button 8, same_screen YES
```

**Macro Example**

```
// Actual: Mouse Button 8 (BACK?)
// holdKey mouseBtn4

// Actual: Mouse Button 9 (FORWARD?)
// holdKey mouseBtn5

// Actual: Mouse Button 10
// holdKey mouseBtn6

// Actual: Mouse Button 11
// holdKey mouseBtn7

// Actual: Mouse Button 12
// holdKey mouseBtn8

// Actual: Mouse Button 13
// holdKey mouseBtn9

// Actual: Mouse Button 24
holdKey mouseBtn20
```

**Working**

Binding a key to mouse scroll up/down seems to work fine, but that can't be done from within a macro so far as I can tell.

```
{
  "keyActionType": "mouse",
  "mouseAction": "scrollUp"
},
```

kareltucek, 2024-04-23:
> Binding a key to mouse scroll up/down seems to work fine, but that can't be done from within a macro so far as I can tell.

What about:

```
startMouse scroll up
delayUntilRelease
stopMouse scroll up
```

----

As for the shift in mouse buttons, I will look into it.

mondalaci, 2024-05-07:
I don't think the output of `xev` correlates with actual mouse button numbers beyond mouse button 3. When using my dedicated mouse:

- scroll up outputs xev button 4
- scroll down outputs xev button 5
- scroll left outputs xev button 6
- scroll right outputs xev button 7

UHK mouse button 4 starts at xev button 8 and the rest follows.

@benedekkupper Do you have any insight about the above?

kareltucek, 2024-05-09:
Indeed, my external mouse buttons 4 and 5 also report 8 and 9 in xev. 

I.e., so far does not look like a bug. 

Is there actually anything broken about your setup? If so, what is it that is broken?

mondalaci, 2024-06-11:
Looks like a non-issue, so closing it. Feel free to follow up nevertheless.

closed 2024-06-11 by mondalaci