UltimateHackingKeyboard/firmware #780: determining hardware configuration of UHK from macros

mhantsch, 2024-06-11:
I want to use the exact same configuration on multiple UHKs, which have different configurations (ISO vs. ANSI layout, modules attached, UHK60 v1 vs. UHK60 v2, possibly running different firmware versions...). I would like to query these data from my macros, so I can adapt some of the behaviour depending on the hardware.

Could there be some system variables that provide that information?

Something like:

```
if $hardware.layout.isoKey == 1 ...         // has an ISO key
if $hardware.version == 2 ...               // UHK v2
if $firmware.version.major >= 2 ...         // runs at least firmware 2.0.0
if $hardware.attachedModule.trackpoint ...  // has a trackpoint module attached
```

kareltucek, 2024-06-11:
What about instead of introducing a complex system of variables, we introduced just a condition that would allow distinguish specific UHK based on its name? Sounds much simpler and yet still more versatile.

As for connected module, there is `ifModuleConnected` already.

mondalaci, 2024-06-11:
@kareltucek Please provide an example smart macro code for the proposed solution.

mhantsch, 2024-06-11:
> What about instead of introducing a complex system of variables, we introduced just a condition that would allow distinguish specific UHK based on its name? Sounds much simpler and yet still more versatile.

Would be very versatile, but maybe too versatile? It would not help with firmware versions, and I would kind of map names to specific features (ISO vs. ANSI, v1 vs. v2) instead of being able to determine each feature directly. I would end up having something like this in the macro code:

```
if regExMatch($uhk.name, /v1/) ...    // it's a v1 UHK
if regExMatch($uhk.name, /ISO/) ...   // it has an ISO key
```
and now I need a regExMatch() function... :smirk: and I need to name my boards something like "My old UHK v1 ISO" and "My new UHK v2 ANSI";

or I use arbitrary names and I need to put that name => feature mapping into the code:

```
// calculate features of this keyboard
if $uhk.name = 'My old UHK' {
  set uhk_version = 1   // is a UHK v1
  set uhk_iso = 1       // and has an ISO key
} else if $uhk.name = 'My new UHK' {
  set uhk_version = 2   // is a UHK v2
  set uhk_iso = 0       // and has no ISO key
}
```

If I shared my configuration for others to use, it would not work out of the box. These other user would need to add their names and how they map to the hardware features on their UHKs.

> As for connected module, there is `ifModuleConnected` already.

Ah, true. I had overlooked that.

kareltucek, 2024-06-11:
No, we are not going to match regexes.

```
ifUhkName myFirstAndBelovedUhk set leds.enabled 1
```

mondalaci, 2024-06-11:
I'm very much against inflating commands, especially for such niche features. Adding namespaced `$hardware.*` variables make far more sense to me.

kareltucek, 2024-06-11:
Well, since a user may want to distinguish two units of the same keyboard, iffing on name is necessary... and that means adding strings (or more precisely, string references) as a variable type.  That's possible, no problem...

So could you (@mondalaci) list the content of the new namespace? Where should name be placed in the structure? `$uhkName`, or `$uhk.name`, or `$hardware.name`?

mondalaci, 2024-06-11:
@kareltucek I think the names originally proposed by @mhantsch are reasonable. Please list the new variables accordingly, along with some examples of the full syntax they'd be used in smart macros.

mondalaci, 2024-06-11:
Thanks, looks good!

kareltucek, 2024-06-11:
No, it does not. Model is missing from the list. It means that furthermore, we will have to have enum values for UHK60v2, ...

kareltucek, 2024-06-11:
Now this sounds reasonable.

mhantsch, 2025-04-09:
Some more suggestions stemming from the forum discussion at https://forum.ultimatehackingkeyboard.com/t/agent-and-firmware-for-uhk60/2271/7?u=maexxx 

```
if $hardware.feature.bluetooth ...
if $hardware.feature.battery ...
if $hardware.feature.backlight ...
```