UltimateHackingKeyboard/firmware #783: [Trackball] Moving the cursor takes a lot of force

Vistaus, 2024-06-13:
I've had the trackball for a while now and while it's overall great, I do have one issue with it: it often takes a lot of force to move the cursor. I've played around with all of the settings, both in the agent as well as in the system settings, but it doesn't make it any easier to move. The only thing that slightly changes it is increasing the base speed, but then cursor becomes far too jumpy and loses all of its accuracy. And even then moving the cursor is only slightly easier, it still takes more force than it should.

I also have different trackballs (ordered them from the UHK shop along with the module) and they all seem to have the same issue. I don't know if this is a software or hardware bug, but it often takes so much force that it feels like as if the ball is there too tightly or something? Maybe that's the best way to describe it.

OS: openSUSE Tumbleweed and Deepin V23 (based on Debian Stable)
DE: KDE Plasma and Deepin Desktop Environment
X/Wayland: Wayland on Plasma, X on Deepin

kareltucek, 2024-06-13:
You mean actual physical force, not speed? If so, which ball are you using? 

Generally heavy matte-finished balls can have that problem, or a badly shaped ball can easily get stuck. 

The light plastic red ball should give good results I believe, although I am using custom made balls 😅.

Also, try to clean the optical piece with a brush (to get any dust out of there), and the bearing balls (just scrape the dirt off with a fingernail).

Vistaus, 2024-06-13:
Thanks for your response. I've already cleaned both the balls and bearing balls several times, but this issue was present from the start (right after the first time using it). It also happens with light plastic red ball, but I don't like the feeling of that one anyway.

A badly shaped ball could be possible, but then it would be on you guys, because these are the official balls. But judging the balls by looking at them, I don't see anything badly shaped about them.

kareltucek, 2024-06-13:
> A badly shaped ball could be possible, but then it would be on you guys

I can't deny that 🤷.

But as far as balls go, the default matte ones (resin black, plastic red, steel) were always without any problem for me in this respect. Prototypes of the glossy one were the only problematic in this sense, not sure if it was resolved before production.

-----

So what happens when you try to move it without pressure? I guess your finger slips?

Jopie01, 2024-06-13:
Just to chime in here. I'm also using the trackball with the steel ball. I have removed the top ring because I have to clean the ball very regularly because I have sweaty hands which make the reaction bad. After cleaning it works again.

Also because the sensor which measures the ball moving is at the underside of the ball it will get some dirt on the lens. A good blow will eventually clean it. What also helps for me is to use a microfiber cloth and put that between your finger and ball and move the ball that way. After doing that everything is working smooth again.

I don't add any force to the ball, just a little bit to get it rolling.

mondalaci, 2024-06-16:
This doesn't seem like a firmware issue and should be better suited as a forum topic, so closing it. Feel free to follow up regardless.

closed 2024-06-16 by mondalaci