UltimateHackingKeyboard/firmware #787: Documentation clean-up

rpnfan, 2024-06-24:
https://github.com/UltimateHackingKeyboard/firmware?tab=readme-ov-file

still lists the old fork from Karel, which is not "needed" any longer, because all functions are now part of the main fork.

The second mentioned custom firmware is also not relevant due more option to configure the SecondRole, right?

kareltucek, 2024-06-26:
> still lists the old fork from Karel, which is not "needed" any longer, because all functions are now part of the main fork.

Indeed.

> The second mentioned custom firmware is also not relevant due more option to configure the SecondRole, right?

Well, general feedback on my implementation of secondary roles is that people still struggle to find a configuration that would not cause spurious activations for them, so I don't consider that feature fully solved on UHK side. I.e., I still consider that alternative firmware to be relevant.

mondalaci, 2024-06-26:
@kareltucek Do you plan to update the readme? Do you plan to take any (other) steps to resolve this issue? I'm asking because I don't want to leave it hanging.

kareltucek, 2024-06-26:
Feel free to do it, I have no strong feelings about it...

mondalaci, 2024-06-26:
How about removing the Custom Firmwares section? I think it doesn't add value anymore with the relevant features present in the official firmware.

mhantsch, 2024-06-26:
> Well, general feedback on my implementation of secondary roles is that people still struggle to find a configuration that would not cause spurious activations for them, so I don't consider that feature fully solved on UHK side. I.e., I still consider that alternative firmware to be relevant.

I am deep into the topic of homerow mods now, and yes, the UHK firmware doesn't get it done well yet _for this particular use case_. Generally I think the advancedStrategy works pretty well for very common use cases.

kareltucek, 2024-06-26:
> How about removing the Custom Firmwares section? I think it doesn't add value anymore with the relevant features present in the official firmware.

Feel free to.

mondalaci, 2024-06-26:
Done.

closed 2024-06-26 by mondalaci