UltimateHackingKeyboard/firmware #788: Per-key tooltip storage

mondalaci, 2024-06-27:
This issue is the firmware side of https://github.com/UltimateHackingKeyboard/agent/issues/2289

@kareltucek Can we use the most significant bit of `key_action_t.type` to store the `hasNote` flag? The note string would be stored as the last element of `key_action_t`. I'm asking to ensure I'm not missing anything.

kareltucek, 2024-06-27:
>  Can we use the most significant bit of key_action_t.type to store the hasNote flag?

I think yes.

> The note string would be stored as the last element of key_action_t. 

We don't want to load it in firmware (into `key_action_t`), do we?

(I don't want to waste another precious 2*12*4*65 bytes on this.)

kareltucek, 2024-06-27:
Then we are on the same page.