UltimateHackingKeyboard/firmware #793: Touchpad clicks+drags when normal mouse move is intended

bhaeussermann, 2024-07-24:
Keyboard: UHK 60 v1
Firmware version: 11.1.1
Operating System: Windows 11 Enterprise

Every now and then when I swipe with my thumb across the touchpad to move the mouse it performs a click-and-drag action instead of just moving the mouse. Unfortunately, I have no reliable way of reproducing the issue. I did check that the touchpad is clean and not grimy.

To be honest, I don't care about being able to drag-and-drop using the touchpad as those interactions typically require delicate movement where I typically use the mouse instead. If there were an option to disable the drag-and-drop capability I would probably opt for that in favor of having to deal with accidentally selecting or dragging things every now and then.

mondalaci, 2024-07-24:
We haven't received such a report yet, and we should address the root cause of the phantom clicks that trigger drag-and-drops.

Please try the relevant points of [this guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys), including the bottom I2cRecovery testing part, and let us know about your findings.

bhaeussermann, 2024-08-08:
Thank you for the advice.
I'm still not able to pinpoint the problem. There don't seem to be any communication errors as the I2cRecovery counter never increments when I enable the I2cRecovery debugging. My keyboard also never misses or duplicates any key strokes.
My most likely explanation at this stage is that the wonky touch-detection might be due to the dry skin on my thumb as we are currently in our dry winter season in the [South African Highveld](https://en.wikipedia.org/wiki/Highveld).

I will report back if I am able to find out more or if I manage to resolve the problem.

mondalaci, 2024-08-08:
This might be an environmental-related problem, indeed. If you can't solve this issue and your touchpad is under warranty, we can replace it, although I'm not sure if it'll solve your issue.

TorC8, 2026-03-04:
I can confirm the problem behavior with recent firmware.

Updated my UHK V1 from (I think) a 1.x to 16.1.1 and immediately started having all kinds of problems with dragging when a normal move was intended.  I won't say it never happened before, but it wasn't enough to be terribly bothersome.

I've been using the touchpad as my primary mouse for probably at least a year since the button on my trackball failed, so it's very nearly certain to be something that changed in the numerous intervening firmware releases.

As for resolution, my first thought had been a drag start sensitivity option, although a disable drag would suffice for my needs.