UltimateHackingKeyboard/firmware #795: Build warnings

mondalaci, 2024-07-31:
@benedekkupper Just upgraded to Linux Mint 22, and I'm getting build warnings. The build works fine, but I'm concerned about these warnings, and unless justified, I want to silence them. Please help. A PR is welcome.

```
$ make
    CC    caret_config.c
    CC    config_manager.c
    CC    debug.c
    CC    eeprom.c
    CC    event_scheduler.c
    CC    i2c.c
    CC    i2c_error_logger.c
    CC    i2c_watchdog.c
    CC    init_clock.c
    CC    init_peripherals.c
    CC    keymap.c
    CC    key_states.c
    CC    layer_stack.c
    CC    layer_switcher.c
    CC    led_display.c
    CC    ledmap.c
    CC    led_pwm.c
    CC    macro_events.c
    CC    macro_recorder.c
    CC    main.c
    CC    module.c
    CC    mouse_controller.c
    CC    mouse_keys.c
    CC    postponer.c
    CC    right_key_matrix.c
    CC    secondary_role_driver.c
    CC    segment_display.c
    CC    slave_scheduler.c
    CC    str_utils.c
    CC    test_switches.c
    CC    timer.c
    CC    usb_composite_device.c
    CC    usb_protocol_handler.c
    CC    usb_report_updater.c
    CC    utils.c
    CC    bus_pal_hardware.c
    CC    command.c
    CC    fsl_rtos_abstraction.c
    CC    basic_types.c
    CC    config_globals.c
    CC    parse_config.c
    CC    parse_keymap.c
    CC    parse_macro.c
    CC    parse_module_config.c
    CC    usb_device_ch9.c
    CC    usb_device_class.c
    CC    usb_device_hid.c
    CC    commands.c
    CC    core.c
    CC    debug_commands.c
    CC    keyid_parser.c
    CC    key_timing.c
    CC    scancode_commands.c
    CC    set_command.c
    CC    shortcut_parser.c
    CC    status_buffer.c
    CC    string_reader.c
    CC    vars.c
    CC    adc.c
    CC    led_driver.c
    CC    merge_sensor.c
    CC    reset_button.c
    CC    test_led.c
    CC    is31fl3xxx_driver.c
    CC    kboot_driver.c
    CC    touchpad_driver.c
    CC    uhk_module_driver.c
    CC    usb_command_apply_config.c
    CC    usb_command_exec_macro_command.c
    CC    usb_command_get_adc_value.c
    CC    usb_command_get_debug_buffer.c
    CC    usb_command_get_device_property.c
    CC    usb_command_get_device_state.c
    CC    usb_command_get_module_property.c
    CC    usb_command_get_slave_i2c_errors.c
    CC    usb_command_get_variable.c
    CC    usb_command_jump_to_module_bootloader.c
    CC    usb_command_launch_eeprom_transfer.c
    CC    usb_command_read_config.c
    CC    usb_command_reenumerate.c
    CC    usb_command_send_kboot_command_to_module.c
    CC    usb_command_set_i2c_baud_rate.c
    CC    usb_command_set_led_pwm_brightness.c
    CC    usb_command_set_test_led.c
    CC    usb_command_set_variable.c
    CC    usb_command_switch_keymap.c
    CC    usb_command_write_config.c
    CC    usb_descriptor_configuration.c
    CC    usb_descriptor_device.c
    CC    usb_descriptor_hid.c
    CC    usb_descriptors_microsoft.c
    CC    usb_descriptor_strings.c
    CC    usb_interface_basic_keyboard.c
    CC    usb_interface_gamepad.c
    CC    usb_interface_generic_hid.c
    CC    usb_interface_media_keyboard.c
    CC    usb_interface_mouse.c
    CC    usb_interface_system_keyboard.c
    CC    composite.c
    CC    hid_bootloader.c
    CC    usb_descriptor.c
    CC    microseconds_pit.c
    CC    serial_packet.c
    CC    wormhole.c
    CC    usb_device_dci.c
    CC    usb_device_khci.c
    CC    usb_osa_bm.c
    CC    system_MK22F51212.c
    CC    fsl_adc16.c
    CC    fsl_clock.c
    CC    fsl_ftm.c
    CC    fsl_gpio.c
    CC    fsl_i2c.c
    CC    fsl_pit.c
    CC    fsl_smc.c
    CC    bool_array_converter.c
    CC    buffer.c
    CC    crc16.c
    CC    key_matrix.c
    CC    key_vector.c
    CC    slave_protocol.c
    AS    startup_MK22F51212.S
    LD    uhk60v2.axf

/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-closer.o): in function `_close_r':
/build/newlib-38V0JC/newlib-4.4.0.20231231/build_nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib/libc/reent/closer.c:47:(.text+0xc): warning: _close is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-closer.o): note: the message above does not take linker garbage collection into account
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-lseekr.o): in function `_lseek_r':
/build/newlib-38V0JC/newlib-4.4.0.20231231/build_nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib/libc/reent/lseekr.c:49:(.text+0x14): warning: _lseek is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-lseekr.o): note: the message above does not take linker garbage collection into account
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-readr.o): in function `_read_r':
/build/newlib-38V0JC/newlib-4.4.0.20231231/build_nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib/libc/reent/readr.c:49:(.text+0x14): warning: _read is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-readr.o): note: the message above does not take linker garbage collection into account
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-writer.o): in function `_write_r':
/build/newlib-38V0JC/newlib-4.4.0.20231231/build_nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib/libc/reent/writer.c:49:(.text+0x14): warning: _write is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg_nano.a(libc_a-writer.o): note: the message above does not take linker garbage collection into account
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:      115020 B       463 KB     24.26%
          m_data:       54613 B        64 KB     83.33%
        m_data_2:       58720 B      65280 B     89.95%
        m_noinit:          16 B        255 B      6.27%

```

benedekkupper, 2024-12-05:
We would need to touch the NXP startup assembly files to be able to fix this properly. Is that something you are interested in?

mondalaci, 2024-12-05:
@benedekkupper Yes, if can be done quickly with a patch.

PR #1289 (2025-08-07): Update kinetis build from makefiles to new McuXpresso SDK toolchain
The latest version of McuXpresso SDK was ported to use west (and cmake), which is a great opportunity to harmonize the build tooling and flows of the nordic and kinetis devices. With this update, the build sources and parameters are specified by cmake, with `IS_NRF_SDK` and `IS_MCUX_SDK` variables present to separate nordic and kinetis selections.

There is an important limitation for the build itself. As both SDKs specify their own west extension commands such as `build` and `flash`, they cannot coexist in the same west works [... cut]
closed 2025-08-07 by mondalaci