UltimateHackingKeyboard/firmware #797: Homerow mods

kareltucek, 2024-08-04:
"Mental" note:

https://forum.ultimatehackingkeyboard.com/t/pause-back-in-the-box/1382/7?u=kareltucek

mhantsch, 2024-08-05:
Another interesting read: https://github.com/urob/zmk-config#timeless-homerow-mods

mhantsch, 2024-08-23:
A bit of brainstorming -- items that would help coding some experimental improvements to home-row mods with smart macros:

- a variable that contains the time (in ms) since the last keystroke, i.e. the time that the keyboard has been idle
- a way to bind primary and secondary scancodes to a key, but still execute a macro when it is pressed
- from such a macro:
  - a way to press/hold/release the (primary) scancode that is bound to $thisKeyId
  - a way to activate the secondary action that is bound to $thisKeyId
- a way to `set` secondary actions from smart macros (e.g. to deactivate/activate homerow mods in certain situations)

I believe the first item (idle timer) is not possible to do at all today.
The others could be worked around with lots of macro code, but it's inconvenient.

mhantsch, 2024-09-02:
My two relevant findings that made HRM work for me:

1. disable during fast typing: https://forum.ultimatehackingkeyboard.com/t/miryoku-and-homerow-mods/723/24?u=maexxx
2. avoid same-hand triggers: https://forum.ultimatehackingkeyboard.com/t/miryoku-and-homerow-mods/723/25?u=maexxx

mhantsch, 2024-10-15:
A little while ago I discussed with @mondalaci some distinct improvements that will allow easy creation of homerow-mods from simple macros:

1. Parametrised macros: allow binding of one single hrm-macro to all the hrm keys, and use parameters to specify the primary scancode, the secondary mod, and possibly a timing parameter that may be different for each finger.
2. Built-in variables containing time since the previous keypress / release (in ms), so the hrm-macro can take different paths based on the interval between the previous keypress and the current one.
3. An additional option to ifSecondary/ifPrimary (`avoidSameHand` or `primaryWhenSameHand`?) that will trigger the primary resolution when a key of the same hand (keyboard half) is pressed, unless the secondary timeout is reached already (then it will use secondary resolution even with keys from the same hand).

Each of these features can be implemented separately, and each make hrm-macros simpler in a specific way:

1. Reduces the number of macros needed, as the same macro can be used for multiple keys.
2. Removes the need for a separate background timer fork, as the hrm-macros can just read the variable. Additionally, it completely eliminates the need to map macros onto all alpha keys; only macros on the hrm-keys are needed.
3. Removes the need to have separate macros for the left and right keyboard halves. In addition, it improves timing and responsiveness because ifShortcut followed by ifSecondary check will be combined into a single command. No need to wait for the timeout of ifShortcut before the ifSecondary check begins.

So it makes sense to even implement just one of the improvements in a firmware release, and follow up later with more.

mhantsch, 2025-08-06:
Note for point 2: [Kanata](https://github.com/jtroo/kanata/) uses [`key-timing` and `key-history`](https://github.com/jtroo/kanata/blob/v1.9.0/docs/config.adoc#key-history-and-key-timing) to refer to the scancodes and timing of previous keypresses. This allows checking up to 8 keys from the past.