UltimateHackingKeyboard/firmware #799: macros should be stopped when saving/loading new configuration

mhantsch, 2024-09-04:
Reset configuration to start fresh.

Create a macro **runner**:

```
setVar variable 0
while ($variable >= 0) {
    setVar variable ( $variable + 1 )
    setLedTxt 555 "$variable"
    delayUntil 444
    ifKeymap DVO break
}
```

Bind to a key. Save config, then hit that key so that runner is running. Watch the display count.

Create a new macro that does something with variables,  eg. I created **`$onInit**:

```
setVar new1 1
setVar new2 2
setVar new3 3
```

Save config. Wait. On my UHK, the display froze, then a little later I got these errors:

```
Error at runner 1/1/8: Too many variables. Can't allocate more than 536935944 variables:
> 1 | [setVar variable 0](https://github.com/UltimateHackingKeyboard/firmware/issues/new#/macro/3?actionIndex=0&lineNr=1&columnNr=8&inlineEdit=true)
>   |        ^
Error at runner 1/2/9: Variable not found: variable
> 2 | [while ($variable >= 0) {](https://github.com/UltimateHackingKeyboard/firmware/issues/new#/macro/3?actionIndex=0&lineNr=2&columnNr=9&inlineEdit=true)
>   |         ^
Error at runner 1/3/12: Too many variables. Can't allocate more than 536935944 variables:
> 3 |    [ setVar variable ( $variable + 1 )](https://github.com/UltimateHackingKeyboard/firmware/issues/new#/macro/3?actionIndex=0&lineNr=3&columnNr=12&inlineEdit=true)
>   |            ^
Error at runner 1/3/24: Variable not found: variable
> 3 |    [ setVar variable ( $variable + 1 )](https://github.com/UltimateHackingKeyboard/firmware/issues/new#/macro/3?actionIndex=0&lineNr=3&columnNr=24&inlineEdit=true)
>   |                        ^
Error at runner 1/4/21: Variable not found: variable"
> 4 |    [ setLedTxt 555 "$variable"](https://github.com/UltimateHackingKeyboard/firmware/issues/new#/macro/3?actionIndex=0&lineNr=4&columnNr=21&inlineEdit=true)
>   |                     ^
Error at runner 1/2/9: Variable not found: variable
> 2 | while ($variable >= 0) {
>   |         ^
E
```

When I tried to reproduce the same steps, I got different errors:

```
Error at God mode in Doom 1/4/21: Variable not found: variable"
> 4 |     setLedTxt 555 "$variable"
Error at God mode in Doom 1/2/9: Variable not found: variable
> 2 | while ($variable >= 0) {
Error at God mode in Doom 1/3/12: Too many variables. Can't allocate more than 536935944 variables:
> 3 |     setVar variable ( $variable + 1 )
Error at God mode in Doom 1/3/24: Variable not found: variable
> 3 |     setVar variable ( $variable + 1 )
Error at God mode in Doom 1/2/9: Variable not found: var
```

Clearly some race conditions happen here. I think the culprit is that macros are not stopped when the configuration is reloaded, which messes up memory.

mhantsch, 2024-09-06:
And memory corruption. `$variable` was never used in the macro `God mode in Doom`.
I've also managed to get the UHK to crash in a way that it needed power cycling to work again. That should never happen on config save!

kareltucek, 2024-09-06:
Yes yes, noted! Will take some time for me to look into it unfortunately.

Thanks for report!