UltimateHackingKeyboard/firmware #805: Addition of more XF86 keyboard symbols

linduxed, 2024-09-14:
Firstly, let me say that I have full respect for the fact that you likely have already considered this, and that you probably have already addressed this in some previous issue (I couldn't find it, but I'm sure it's here somewhere).

I was adding some keys and macros to a utility layer, and I was reminded of the `XF86Sleep` key that is available. I then noticed that there's no key for just initiating the screensaver/locking. I googled a bit and found [this](https://wiki.linuxquestions.org/wiki/XF86_keyboard_symbols) list of XF86 keys, and there seems to exist such a notion as `XF86ScreenSaver`. I'm not sure how common this key is, but it would have found it handy to add to my layer, to then configure my system to lock the screen if it catches this key getting pressed.

I then realized that I neither know what keycode/keysym (this is the terminology used by `xev`) is used for this `XF86ScreenSaver` thing, and further that there's currently no way to enter arbitrary scancodes in UHK Agent.

I could probably just make use of some other scancode that I'll likely not use for its named purpose ("Eject" and "Launch Email Client" seem like prime candidates), but it got me generally curious about the following questions:

1. Do all of these various XF86 keys correspond to scancodes?
2. Is the current set of available scancodes/keys just the set of most common ones?

Also: thanks for all the work of the UHK team and all of the contributors!

kareltucek, 2024-09-14:
> and further that there's currently no way to enter arbitrary scancodes in UHK Agent.

This is not correct. See https://ultimatehackingkeyboard.com/blog/2021/10/31/production-progress-spare-parts-bonanza-new-agent-release#custom-scancodes

As to your questions:

1. I don't think so

2. I believe yes

but I am not sure about either of them.

linduxed, 2024-09-15:
Thank you for that correction! I was convinced that my experimenting with the UHK interface suggested that this functionality was not there, but I'm happy to hear that I was wrong!

Also, thanks for your answers to the two questions.

mondalaci, 2024-09-15:
I usually use F13-F24 and combine them with [multiple] modifiers to create shortcuts that don't conflict with others.

Closing this issue, but feel free to follow up.

closed 2024-09-15 by mondalaci