UltimateHackingKeyboard/firmware #812: Updating Firmware to 11.2.0 always fails

Warrost, 2024-10-21:
Before submitting a new issue, make sure to do the following:

1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
3. Our contact form will suggest you solutions, so copy-paste your message at https://ultimatehackingkeyboard.com/contact-us - Please don't post customer support issues here.

Latest Agent is installed, trying to update to 11.2.0 via firmware bundled with agent, nor manually downloading the firmware and using the agent to flash it works.
Things I did:

- Changed USB-C cable
- unplugged modules
- directly connecting both halves

Nothing works.
However flashing back to 11.1.1 or 11.1.0 works without an issue.

[Start flashing firmware.txt](https://github.com/user-attachments/files/17461919/Start.flashing.firmware.txt)

mondalaci, 2024-10-21:
Please retry on another computer.

Warrost, 2024-10-21:
> Please retry on another computer.

Hi there,

I've used a different PC, but it still doesn't work.
Same issue.

Best regards.

mondalaci, 2024-10-21:
@ert78gb Please look into this.

ert78gb, 2024-10-21:
based on the logs. The right half can't communicate with the left half. `[DeviceOperation] Cannot ping the bootloader. Please remove the "Left keyboard half" module, and keep reconnecting it until you see this message.`

Please start the firmware upgrade and check the logs. If you see the `[DeviceOperation] Cannot ping the bootloader. Please remove the "Left keyboard half" module, and keep reconnecting it until you see this message.` message then please disconnect the bridge cable from the left half and reconnect again. Wait few seconds and if you still see the mentioned error message then repeat the steps.

ert78gb, 2024-10-22:
Hi @Warrost ,

Could you run the Agent with `-- --log=misc,usb` command line arguments and send me the logs.

Thanks,
Robi

Warrost, 2024-10-22:
Hi @ert78gb ,

[log-misc_USB.txt](https://github.com/user-attachments/files/17473251/log-misc_USB.txt)

Best regards

ert78gb, 2024-10-22:
@Warrost the attached log file does not contain the usb communication entries. the two times of the double dash `--` is not a mistake. You have to use them. After you started the Agent please also run the firmware upgrade process.

Warrost, 2024-10-23:
@ert78gb ahh got you, I attached the correct log now.  
[Uploading log-misc_USB.txt…]()

mondalaci, 2024-10-23:
@Warrost It's still worth reconnecting the halves until the firmware update works. Sometimes, it takes a lot of tries. Alternatively, feel free to [contact us](https://ultimatehackingkeyboard.com/contact-us), and we'll recall your UHK.

mhantsch, 2024-10-23:
Btw, the message `[DeviceOperation] Cannot ping the bootloader. Please remove the "Left keyboard half" module, and keep reconnecting it until you see this message.` is confusing. I guess you mean "keep reconnecting it until you do NOT see this message anymore"?

Warrost, 2024-10-23:
> @Warrost It's still worth reconnecting the halves until the firmware update works. Sometimes, it takes a lot of tries. Alternatively, feel free to [contact us](https://ultimatehackingkeyboard.com/contact-us), and we'll recall your UHK.

Thanks for the offer, I'd like to keep my board as long as possible, since I customized it quite a bit.
I'm just confused, since only this particular firmware version leads to issues.  
I have no problem whatsoever when I flash 11.1.1 or 11.1.0.  
Only 11.2.0 seem to have a problem with the bootloader.  
I also have no problems in the daily operation of the keyboard, it works perfectly fine.

mondalaci, 2024-10-23:
@mhantsch Great catch, thanks!

@ert78gb Please update the message accordingly.

@Warrost Your finding is very unusual. Can you consistently flash 11.1.1 and 11.1.0 but not 11.2.0?

ert78gb, 2024-10-24:
> Only 11.2.0 seem to have a problem with the bootloader.

The checksum of the installed left firmware version (10.3.0) and the 11.1.1 or 11.1.0 are equal, so Agent skips the firmware upgrade of the left half.

closed 2025-09-16 by kareltucek