UltimateHackingKeyboard/firmware #813: Something changes the `deviceProtocolVersion`

ert78gb, 2024-10-23:
A few customers reported Agent does not recognise the keyboard. 
- https://github.com/UltimateHackingKeyboard/agent/issues/2284
- https://github.com/UltimateHackingKeyboard/agent/issues/2273

After investigated the logs I recognised the `deviceProtocolVersion` has invalid value like `1280.0.0` or `512.0.0`
The `GetDeviceState` also returns with invalid `XmoduleSlot` identifier.

If the user flash the right half firmware then everything works again.

My assumption is something shift the data in the firmware. I can't prove it.
It is very rare case and we can't reproduce. I asked the users to share their user config and now waiting.

ert78gb, 2024-10-23:
I forgot the powercycle option, so I did not ask them but I will try next time

kareltucek, 2025-09-16:
Closing since we haven't seen it in a while.