UltimateHackingKeyboard/firmware #823: ISO keys are lit in switch tester mode on the UHK 80

mondalaci, 2024-12-10:
When the layout is set to ANSI on the UHK 80, and the switch test mode is enabled, I press a key. Only this key should be lit but the ISO Shift and the extra ISO key are also lit.

mondalaci, 2024-12-10:
@kareltucek Please make this a priority.

kareltucek, 2024-12-10:
I am not sure how to reproduce this.

> I press a key

You press which key?

When hwconfig on my uhk80 is set to ansi layout and I press ansi shift, only ansi shift get's lit.

mondalaci, 2024-12-10:
1. The layout is ANSI.
2. I enable the switch testing mode.
3. I press escape, and only it should be lit. I see this instead:
[image]

kareltucek, 2024-12-10:
Is user config flashes?

mhantsch, 2024-12-10:
> only ansi shift get's lit.

@kareltucek Are you testing on an ISO or an ANSI UHK 80?

kareltucek, 2024-12-10:
I am testing with ansi board, but can see the iso leds quite clearly.

PR #825 (2024-12-11): Turn off all led slots again when the light all mode is exited
Maybe fixes #823
closed 2024-12-11 by mondalaci