UltimateHackingKeyboard/firmware #1000: UHK 80 text macro action mistypings

mondalaci, 2024-12-12:
* Sometimes, instead of `forum.ultimatehackingkeyboard.com`, `forum.ultimaethackignkeyboard.cmo` is outputted.
* rollbar.com -> rolblar.com
* x.com/notifications -> xc.om/notifications

pcooke9, 2024-12-14:
Thought I'd mention this just happened to me again for the first time in quite a while on the UHK80.

•keystroke delay = 10ms
•firmware-uhk80: 11.2.0 #stuck_keys_hunt_v4
•both halves connected to the PC via USB, and spiral bridge cable connected as well.

mondalaci, 2024-12-26:
@pcooke9 Please retest this issue with firmware 12.1.0. I can't reproduce it anymore.

pcooke9, 2024-12-26:
> @pcooke9 Please retest this issue with firmware 12.1.0. I can't reproduce it anymore.

I haven't had it happen since my previous comment, but it always was a rare occurrence for me. I've been keeping up to date with the firmware, and I'll comment here if I see it again.

mondalaci, 2025-01-08:
Based on my experience, this has been resolved, so I'm closing it.

closed 2025-01-08 by mondalaci