UltimateHackingKeyboard/firmware #1004: Undetected and chattering key presses

mondalaci, 2024-12-12:
@jwr reported:

> I am pretty sure there is a rare problem with keypresses not being detected or being detected multiple times. Debouncing code might need improvement.

I've also noticed this daily, but I wasn't sure until now. Maybe the key scanner thread should be given higher priority.

mondalaci, 2025-01-08:
This is not an issue for me anymore.

closed 2025-01-08 by mondalaci

jwr, 2025-01-08:
Hmm. This is still very much an issue for me, but I'm using 12.0.1 (for example, typing this, I just had to delete an additional '1' that appeared at the end of that version number, and then also correct the word 'appeareed'). And I can't use anything past 12.0.1. So, it's difficult for me to confirm that the issue has been fixed.

mondalaci, 2025-01-09:
Don't you think this issue is a duplicate of #1077?

jwr, 2025-01-09:
No, I don't think so. I had problems described in this issue for a long time now, and I still have them with 12.0.1. These problems are an annoyance, but they don't make the keyboard unusable. Also, I never saw swapped keystrokes (not in order) with 12.0.1.

What is described in #1077 started happening (in my case) after 12.0.1, and is much more severe.

That said, perhaps the code has been reworked so much that this issue no longer applies at all, but I wouldn't know that.

mhantsch, 2025-01-09:
I agree to what @jwr said. 12.0.1 has some (occasional) repeats, but is reasonably usable. Later versions were unusable for me; I couldn't type a single meaningful sentence.

spirodonfl, 2026-01-31:
I also have this issue and, while it's mostly consistent on which keys it affects, it does sometimes change very rarely. For the most part, the SHIFT button is not detected when I hold it down (noticeable when I'm programming). The r key is often repeated twice. Previous firmware versions had issues with the w key being repeated (no longer seems a problem) or the d key being repeated multiple times. So, while there does seem to be an improvement, it does still exist for me.

mhantsch, 2026-02-01:
Some things to check:

Are your switches in sockets (UHK60v2, UHK80) or soldered (UHK60v1)? If in sockets, you can try removing those that play up, bending their pins slightly and putting them back in the socket.

Especially for UHK60: Are only keys on the left half affected? You might have interference on the (spiral) bridge cable between the halves. 

Generally, EMV interference can cause problems with the UHK60 (especially v1). The first sets of UHK60v1 had grounding issues, see here: https://uhk.io/repair-and-modding/esd-protection-guide

There's a full guide to troubleshooting key chatter here: https://uhk.io/repair-and-modding/troubleshooting-repeated-and-missing-keystrokes

kareltucek, 2026-02-03:
We are indeed having some reports about some i2c reliability issues on 16.1.x, so trying 16.0.x may be a good think if you suspect software cause. 

@makenomistakes are right half keys affected too, or only left half plus modules?

kareltucek, 2026-10-23:
Another iteration of hunting done in 18.0.0. I think we should be now rock solid.

Can you guys confirm?