UltimateHackingKeyboard/firmware #1006: Hang/freeze report

mondalaci, 2024-12-12:
Reported by @jwr:

>I woke up my Mac this morning and found the UHK80 unresponsive. No keypresses were sent to the Mac. The keyboard did wake up when I pressed keys, the OLED was showing "Mac OS USB cable" and a wireless symbol. Right half was connected with USB, left half was not connected at all (no spiral cable). After disconnecting USB and reconnecting to the host, the right half started responding. It prompted me for a pairing code, so I pressed some numbers to get "pairing failed". But the left half was still unresponsive. It did something, because I was able to press the combination for switching keymaps and the OLED updated, but no keypresses were sent to the host. I connected the left half with a spiral cable, which changed the symbol on the OLED to a plug, but no change: keypresses were still not being sent. What finally helped was the RESET button in the left half, which restored the keyboard to normal functionality. This is with firmware 5aaef90.

@kareltucek If you have a hunch, it may be worth inserting debugging statements that display specific texts on the OLED display on relevant occasions.

jwr, 2024-12-12:
This sort of hang/freeze happens to me regularly. Usually after I wake my Mac from sleep, but also sometimes during normal usage. I think we can rule out BLE, as I'm not using Bluetooth at all and somehow even managed to get rid of those "pairing" messages.

The symptoms are:

* unresponsive keyboard
* hitting keys *sometimes* results in the keyboard restarting (screen goes blank for a second or two, then comes back and I start getting keystrokes)
* sometimes I have to disconnect the keyboard and connect it again

This happens about once or twice per day.

mhantsch, 2024-12-12:
Seeing the same on UHK 80. Sometimes it doesn't want to wake up - even if connected via USB. OLED screen stays black, it doesn't send any keys, and backlight stays off. Have to hit keys, and after a while it wakes up. Or replug USB. Or restart computer. Any of these will eventually wake up the keyboard.

pcooke9, 2024-12-13:
I occasionally see this happen as well. Tapping random keys restarts the UHK80 and resumes normal operation.

OS is Win 11.

mondalaci, 2024-12-13:
Thanks for your reports, guys! I haven't experienced this yet, but I think this issue will be resolved by https://github.com/UltimateHackingKeyboard/firmware-uhk80/pull/353