UltimateHackingKeyboard/firmware #1007: Battery display changing and then device crash

Zetaphor, 2024-12-12:
I was working on my keymaps in the Agent on Windows when I noticed this happening. I'm not sure if this was happening before I had already launched Agent and saved some changes or after.

http://share.zetaphor.com/u/8ckY3x.mp4

I am connected via USB and bridge cable.

Shortly after this I left the room for a minute and when I came back the keyboard was showing as disconnected. When I removed and reconnected the USB cable Windows error'd and told me it didn't recognize the device. I ended up having to restart both halves.

tunney, 2025-03-06:
Also seeing this one. Yesterday I had the L and R being displayed. This was as the sides were charging.
Left plugged in over night I only see R100+ — I am wondering is it a result of both sides being fully charged?

I have the keyboard connected to the host on usb
and when the sides are connected with both fully charged I only see R100+

But as soon as I disconnect the bridge cable I see "100- 100+" so I think it is definitely related to being fully charged, as least for me.

kareltucek, 2025-03-06:
I  think that it has to do with the connected bridge cable which for some reason messes up battery presence detection.

kareltucek, 2025-09-16:
If you are still experiencing similar issues, please open please a new ticket ;-).