UltimateHackingKeyboard/firmware #1009: Mouse key scroll inconsistently firing

Zetaphor, 2024-12-12:
I'm not sure if this is connected to a related issue, but I've noticed that there is inconsistency in the signal sent from the mouse scroll down key  (using mouse keys). This came up from subjective experience as I rely heavily on the mouse layer, then I tested it with some online key testers.

When I test the mouse down key here:
https://cps-check.com/mouse-buttons-test

I can see that while I keep the physical key pressed the signal the page receives is intermittent, but not at a regular frequency. Sometimes it stays held, but it mostly switches on and off.

However when I use the Page Down key, which is on the same layer on this page:
https://keyboard-test.space/

The key is held consistently and is only released when I physically release it.

Here's the mapping I'm using for this key in the Agent:

[image]

I'm encountering this issue on both the USB dongle and USB cable

kareltucek, 2024-12-27:
@Zetaphor is this still relevant?

Zetaphor, 2024-12-29:
It appears this issue has been resolved with the latest firmware.

closed 2024-12-29 by Zetaphor