UltimateHackingKeyboard/firmware #1016: UHK80 Failing USB Communication

kareltucek, 2024-12-12:
[image]
[image]
[log.txt](https://github.com/user-attachments/files/17376522/log.txt)
[log2.txt](https://github.com/user-attachments/files/17376523/log2.txt)

TL;DR:

Firmware to Agent communication sometimes seems to miss a packet.

I haven't investigated it much though.

kareltucek, 2025-04-04:
This seems to be happening when I resize Agent window while configuration is loading. 

I use a tiling window manager, so I usually hit a shortcut that moves the window right after starting Agent, which is why I am frequently affected but many users are not.

@ert78gb 

@mondalaci would you transfer the issue to Agent repository? I either don't know how to do it or don't have permissions 😃.

ert78gb, 2025-04-04:
Really surprising me but I will investigate.