UltimateHackingKeyboard/firmware #1017: UHK80 requires disconnecting and reconnecting after suspend

jwr, 2024-12-12:
Sometimes, when my computer wakes up from sleep, the keyboard will not send any keystrokes to the computer. It appears to be on, and the display says "USB cable," but no keystrokes are being received by the Mac. I think this problem began after upgrading to c75c97f.

Disconnecting the USB cable and reconnecting it again makes the keyboard work.

benedekkupper, 2025-04-22:
Those who are affected please test this firmware (current master branch): https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14422644330/artifacts/2933383354

mondalaci, 2025-05-04:
@jwr Can you test it?

jwr, 2025-05-05:
It has only been a couple of days since I upgraded (I stayed on 12.0.3 for a very long time, because at least the bugs were known and I needed to get work done), but so far 👍 — I can wake my Mac using the keyboard, and it doesn't reset itself after wakeup.

Please give me several more days to confirm with more wakeups, but so far it looks good!

jwr, 2025-05-09:
After a couple more days, I can confirm that this firmware is definitely better than 12.0.3. I have not seen a single wakeup problem since I upgraded. Definitely 👍

mondalaci, 2025-05-09:
Thanks for confirming! Closing this issue.

closed 2025-05-09 by mondalaci

mhantsch, 2025-05-09:
For documentation purposes: I did have some wakeup issues with the UHK80 over USB with firmware 13.0.2 still, but I have since upgraded to 14.0.0 and that seems to be working fine so far.