UltimateHackingKeyboard/firmware #1018: Port hi-res scrolling to Zephyr/c2usb

mondalaci, 2024-12-12:
Based on https://github.com/UltimateHackingKeyboard/firmware/pull/794

benedekkupper, 2024-12-18:
What's the most straightforward way to test this?

mondalaci, 2024-12-18:
@benedekkupper Please let me know your OS.

@mhantsch @rightaditya Would you provide some instructions? Benedek is about to port hi-res scrolling to the UHK 80. I'm wondering whether it's too early to port the exact descriptors and report format given #1039.

rightaditya, 2024-12-19:
Testing should be straightforward. I've only tested using the trackball and mouse scroll keys (@mhantsch said it worked for the touchpad too), but if you compare scrolling with the patch vs. without, it should be immediately obvious, so long as the application supports it. Windows is probably best for this given the variance between apps and installations we've found on Linux (and, IIRC, macOS also varied between users). On Windows I found that loading any website in Edge that required scrolling to be suitable as a test. Horizontal scrolling is harder to find but searching for HTML horizontal scroll test pages gave me something I could use.

It's probably best to use the default speed etc. settings when trying this, but tbh the increase in resolution is such that it should be obvious except maybe for very extreme settings (so extreme that I would imagine them to be useless).

So basically:
1. Set trackball module speed settings to their defaults.
2. Open Edge and browse to a suitable (scrollable) page.
3. Scroll with the trackball using the old firmware.
4. Flash the patched firmware or else switch to another keyboard with the patched firmware and scroll again. It should feel obviously smoother.

Scroll keys should show a difference too, but IMO it's not as easily noticed and you have to pay closer attention to the scrolling to see that it's indeed smoother, especially if the application has animated scrolling.

As for porting the descriptors and report format, I don't think anything in #1039 will affect them specifically. FWICT the solution for that issue will be to either fudge the multiplier for one or more of the scroll speed settings or else recommend that users change their values accordingly (I think just changing the initial speed would do it), neither of which affects the HID report or descriptors.

mondalaci, 2025-01-16:
Does #1047 conclude this issue?

mondalaci, 2025-01-16:
Closing this issue, assuming it's resolved.

closed 2025-01-16 by mondalaci