UltimateHackingKeyboard/firmware #1021: Report routing revisited

kareltucek, 2024-12-12:
@benedekkupper, we would like to have more control over the bt/usb HID interfaces. 

Until now, whenever an unknown bluetooth device connected, we have (very naively) assumed it is a HID device, and disabled USB HID, and consequently, have treated the connection states as mutually exclusive. This doesn't fit the bill for muliple reasons.

- we would like to know the connection states independently (i.e., call the DeviceState_SetConnection in `void hidmgr_set_transport(const hid::transport* tp)` less naively)
- when both are connected, we would like to be able to decide where a report is to be sent (basically implementing a kvm switch) (e.g., something like `mouse_app.selec_destination(bt / usb);` would fit the bill nicely.)
- specifically, if I am not mistaken, in certain situations, we would like to send the commands app reports over USB, while sending keyboard app reports over bluetooth

(Do we already have such ability? If now, what course of action do you suggest?)

kareltucek, 2024-12-26:
@benedekkupper what is the state of affairs? 

From my point of view, current blockers are:

- reliable way to tell c2usb to switch to usb or ble hid when both are connected. 
  - I am still not sure what the USB_EnableHid/USB_DisableHid is exactly supposed to do - whether to switch  between usb hid and ble hid operations, or something else? Would you either elaborate (best in code comment), or refactor it somehow?
  - In any case, when Ble Hid is connected and I call USB_EnableHid, the firmware tends to crash, so I *guess* I am not supposed to do that while hogp is active.
- When ble hid is connected, what is the recommended way to tell whether usb hid is available? 
- `void hidmgr_set_transport(const hid::transport *tp)` is unreliable. Just like in the #1060 it often ends up in NULL getting reported as the last state even though usb connection is available.
- the power event delegate isn't reliable. If I am not mistaken, this is a regression after the huge 2.8.0 NCS PR.

Other less important thoughts - if you have any comment, it is highly welcome:

- I am not sure how c2usb actually takes over the ble hid connections. I would expect the bt_conn.c code to explicitly call c2usb, telling it "here you have a ble connection handle, please establish hid over it". 
- It would be nice to actually check somewhere that the device that connects is asking for ble hid rather than nus. At the moment the code assumes that if an address is not known, then it is ble hid. (This doesn't generate any troubles at the moment as far as I know, but I am uneasy about it.)

benedekkupper, 2025-01-08:
What I implemented in the previous [PR](https://github.com/UltimateHackingKeyboard/firmware80/pull/353) was only the concurrent agent operation over USB and BLE. Both can be used at the same time, since it's fully a request-response protocol, the response is always sent where the request comes from.

@kareltucek Now on the `parallel-host-routing` branch I have added the changes [I proposed earlier](https://github.com/UltimateHackingKeyboard/firmware/issues/1021#issuecomment-2537924288), where both USB and BLE HID are live in parallel, and the UHK itself needs to make the decision which way to go. This is WIP, since the actual decision whether to send via `usb_handle` or `ble_handle` needs to be made by the routing logic - see the **TODOs** I made in `usb_compatibility.cpp`. Besides that, the keyboard LED management also needs to track the USB and BLE host states in parallel independently, so `KeyboardLedsState` needs to be redesigned to accommodate this. (Also note that I moved `Connections_SetState` calls to `keyboard_app.cpp`, please review it.)

closed 2025-01-10 by mondalaci