UltimateHackingKeyboard/firmware #1024: Macro suppressMods bug

kareltucek, 2024-12-12:
https://forum.ultimatehackingkeyboard.com/t/can-i-change-secondary-role-according-to-the-second-key-pressed/1783/10?u=kareltucek

kareltucek, 2024-12-26:
@mhantsch I am quite lost. Would you provide a minimal setup that exhibits the issue? (I.e., a UserConfig.)

mhantsch, 2024-12-27:
> @mhantsch I am quite lost. Would you provide a minimal setup that exhibits the issue? (I.e., a UserConfig.)

Did you try to configure what's described in https://forum.ultimatehackingkeyboard.com/t/can-i-change-secondary-role-according-to-the-second-key-pressed/1783/4?u=maexxx ?

I'm away from my UHK (AFUHK :)) until a few days into the New Year, so I can't access my configuration at the moment.

kareltucek, 2024-12-27:
I had tried a few simple experiments resembling that setup and didn't notice anything odd.

As for the rest of the scenario, I really prefer to wait for you to get back and provide a reasonably-reproducible configuration. If you manage to minimize it somewhere in between, all the better.