UltimateHackingKeyboard/firmware #974: Charging status and percentage

mondalaci, 2024-12-12:
The charging status and percentage of both halves must be displayed on the OLED display.

### Charging status

There are 3 charging statuses: charging, not charging, and battery unavailable (as the UHK 80 will be available without batteries via the webshop).

`chargerStatCallback()` should be used to update the charging state. if `gpio_pin_get_dt(&chargerStatDt)` is 0, charging is in progress; otherwise, the keyboard half is not being charged.

If the battery is absent, `chargerStatCallback()` is called twice per second with `gpio_pin_get_dt(&chargerStatDt)` alternating between 0 and 1.

~The right half has a [hardware bug I'm trying to fix](https://github.com/UltimateHackingKeyboard/firmware/issues/852) that causes the `chargerStatCallback()` to get called gazillions of times per second with a 0 stat value. Let's be aware of this bug, but don't work around it, as I'll fix it on the hardware level.~

Let's keep charging (chargerEnDt) always enabled for now. Later, we can implement a battery conservation mode by disabling charging above 70 or 80 percent.

### Charging percentage

See `cmd_uhk_charger()` for the relevant code.

1. This is the raw ADC value of battery voltage:

```
        err = adc_read(adc_channel.dev, &sequence);
        int32_t val_mv = (int32_t)buf;
```

Measure and monitor this value in a thread at a reasonable interval, such as 5 or 10 seconds. Whenever it changes, make the left half send it to the right half. The right half should process the raw ADC battery measurements as follows.

2. Convert the raw ADC measurement to mV using adc_raw_to_millivolts_dt()

3. The mV value must be converted to percentage, which is a logarithmic-shaped curve. I've made several measurements with the `scripts/log-battery.mjs` script while [dis]charging the battery in `scripts/battery-stats`.

Some wise words from Martin, our electronics consultant:

> Log it out your USB port, and let it run from fully charged to discharged.
> In addition to measuring voltage, it is a good idea to periodically check the current also.
> If current is steady over the full time duration, the # of mA times # of hours should be close to the mAhr capacity of your battery.

If you want to make battery logs, I suggest charging the halves independently with a decent charger that can output at least 3A. In practice, the charging rate will not exceed 2.3A even at the beginning of charging. Alternatively, you can charge one half via the other half, but it'll be slower. A (or two) battery charging meter can come in handy, and I recommend the Fnirsi FNB58 by far. We can purchase one or two, or you can purchase, and we can reimburse you via your next invoice.

Ignore the TS (thermistor) values. We don't want to measure battery temperature via its thermistor anymore, as it wouldn't be helpful and would require additional parts. I've removed the TS ADC node from the devicetree and from the code, but it's still present in `scripts/log-battery.mjs`.

As you can see, the measured voltage slightly fluctuates. We must combat this fluctuation, as it would confuse the user if it were visible.

As for converting the battery voltage to percentage, I think we should use an array to map raw ADC values starting from raw ADC min (indexed as 0) to raw ADC max, mapped to the percentage, based on the stats. This map would differ per keyboard half, given that their batteries have different capacities.

Now that I think about it, the mV value is irrelevant to the above solution as the raw ADC value would directly map to percentages, but it's worth being aware that the raw ADC value is not mV.

### Bluetooth battery percentage reporting

The UHK should report the charging level to the host via Bluetooth so that operating systems can display it. I'm unfamiliar with the details, but Google for `bas_notify()`. Ideally, the UHK could expose the charging percentage of both halves, but I'm not sure whether Bluetooth and the OSes support it. If it's only possible to expose a single percentage value, let's expose the minimum of both.

kareltucek, 2025-05-28:
Can this be closed?

mondalaci, 2025-05-28:
Yes. Closing it.

closed 2025-05-28 by mondalaci