UltimateHackingKeyboard/firmware #976: Minimize audible noise

mondalaci, 2024-12-12:
First, let me emphasize that we should wait with this issue because the final product's sound signature will probably differ from the prototypes.

In `IS31FL3743B_DS.pdf` of the datasheets repo, see the "phase delay" and "spread spectrum" options. There are 2 (PHC) * (4 (RNG) * 4 (CLT) + 1 (SSP=0) ) = 34 options according to the datasheet. I tried these settings a while back, and they have made a big difference, and the settings I found to be best are currently implemented in `ledUpdater()` but they should be re-tested for the final product.

`AN-154-EN IS31FL3743B PWM Frequency Options Rev.A.pdf` features the PWM Frequency setting, which has 7 options. I haven't tried this setting yet.

The optimal settings may differ per keyboard half.

lightningRalf, 2024-12-31:
So as a happy user of the final product I can say the following:

It hums annoyingly, when all the keys are backlit.
The reason I say this is, because the hummings stops when I push down different modifier keys.

With different levels of intensity depending on the number of active backlit leds.
And when putting the level of the backlit LEDs around 40 it is also not hearable for me anymore.

closed 2025-01-01 by mondalaci

mondalaci, 2025-01-01:
Please try the newly released 12.2.0 firmware, which should reduce audible noise, and let me know your findings.

mondalaci, 2025-01-02:
After building Agent, audible noise settings can be adjusted with `/packages/usb/send-command.ts 19 6 phaseDelay spreadSpectrum pwmFrequency` where
- `phaseDelay` can be `0` and `128`,
- `spreadSpectrum` can be `0` and `16` to `31`, and
- `pwmFrequency` can be `0`, `32`, `64`, `96`, `128`, `160`, `192`, `224`, `256`.

These settings don't persist across firmware reboots. According to my observations, only the `spreadSpectrum` setting affects audible noise.

Related documentation:
- https://github.com/UltimateHackingKeyboard/datasheets/blob/e3da58d81e473778f076fadb00bf6940c55d45ed/leds/IS31FL3743B_DS.pdf
- https://github.com/UltimateHackingKeyboard/datasheets/blob/e3da58d81e473778f076fadb00bf6940c55d45ed/leds/AN-154-EN%20IS31FL3743B%20PWM%20Frequency%20Options%20Rev.A.pdf

nref, 2025-01-18:
Hello. Just received my UHK 80 with firmare 12.2.0. I'm also experiencing this hum when all of the lights are on. While I press a modifier key and the lights are off, the hum does not happen. Thanks.

mondalaci, 2025-01-18:
Hi there! That's because the sum of the light levels is proportional to the noise level. Reducing the brightness level in Agent can dramatically decrease the noise level.

nref, 2025-01-18:
Thanks for the quick response. I can confirm that reducing the brightness (which I was glad to do anyway) reduced the noise level. It's still audible, but hopefully bearable.

Awesome keyboard otherwise!

PR #1071 (2025-01-01): Reduce audible noise, and expose related settings for experimentation…
… via the device protocol.

Closes #976.