UltimateHackingKeyboard/firmware #980: BLE HID pairing

mondalaci, 2024-12-12:
BLE HID pairing already works to some extent. The pairing code appears on the OLED display, it's accepted/refused, and the relevant security keys are created on the settings partition.

However, the paired device should also be added to host connections, which is not trivial because it changes the user configuration, which should be updated by Agent.

1. The firmware calculates **new addresses** by reading the pairing keys on the settings partition (which I believe are of the `bt/keys/e89c2557115f0` format, but double-check it), and checking whether they're present in host connections.
2. If there are new addresses, expose a new `newBleAddressesPresent` bit via getDeviceState's response.
3. When Agent detects this bit, it'll call the newly added `getNewBleAddresses` getDeviceProperty command. The first response byte is the number of new BLE addresses, followed by the 6-byte addresses. Assuming a maximum 64-byte packet with a return status byte and a count byte, only 62 bytes are available for addresses, allowing up to 10 addresses, so maximize the number of addresses in 10. This process will naturally repeat for the rest of the addresses on the rare occasion of more unpaired addresses.
4. Agent will display a yellow notification bar at the top containing "A new Bluetooth device has been paired to this UHK. Let's [add it to connections] so that you can switch to it." or its plural form "New Bluetooth devices have been paired to this UHK. Let's [add them to connections] so that you can switch to them."
5. When clicking on the above [add[s] it/them to connections] link, Agent jumps to the connections screen, adds the new addresses to connection targets, and saves the configuration. The firmware should expose the updated values, making the notification bar disappear.

ert78gb, 2024-12-22:
Tested with the https://github.com/UltimateHackingKeyboard/firmware/commit/711b6e0c88ee018dec02d7a1b42111e128e5af6c

I found the following issues

1) The device state command always set `NewPairedDevice` flag. 
```
❯ ./packages/usb/send-command.ts --log=usb 9

[UhkHidDevice] USB[W]: 04 09 
[UhkHidDevice] USB[R]: 04 00 00 04 01 00 00 00 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Byte result:
0 0 4 1 0 0 0 1 5 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

The new pairings USB property returns correctly with 0 item after the user configuration saved.

```
❯ ./packages/usb/send-command.ts --log=usb 0 12 0

[UhkHidDevice] USB[W]: 04 00 0c 00 
[UhkHidDevice] USB[R]: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Byte result:
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

```

2) The new pairings USB property returns with a BLE address as new pairing even if it is the host connections.
Reproduction
- Remove all the BLE addresses from the host connections
- Pair the keyboard with a BLE device
- Open Agent (that refers to this PR https://github.com/UltimateHackingKeyboard/agent/pull/2450)
- Add the new BLE Address to the host connection
- Close Agent
- Delete the pairing on BLE device
- Wait a little bit 
- Pair again the keyboard to the BLE device
- Open Agent
- Add the new BLE Address to the host connection
- The same BLE address duplicated because Agent current does not check the BLE address exists in the host connections or not. Because my expectation (if I read well the specification) the keyboard should report the BLE addresses that are not in the host connections.

kareltucek, 2024-12-22:
@ert78gb Should be fixed on https://github.com/UltimateHackingKeyboard/firmware/tree/host-routing branch. Please use that one for further pairing tests.

closed 2025-01-10 by mondalaci