UltimateHackingKeyboard/firmware #990: macOS: Immediate reconnection after disconnect

mondalaci, 2024-12-12:
On macOS, clicking on the Disconnect button for the UHK 80 in the Bluetooth device list disconnects it but then immediately reconnects:

```
Disconnected from unknown (10:b5:88:72:9a:e6), reason 19
NUS and HID advertising successfully started
I: stopping HID app 1
W: Spurious suspend/resume event
I: BAS Notifications enabled
Connected to unknown (10:b5:88:72:9a:e6)
Security changed: unknown (10:b5:88:72:9a:e6), level 4
W: Ignoring data for unknown channel ID 0x003a
I: starting HID app 1: 1
BLE HID conn params: interval=15 ms, latency=4, timeout=1000 ms
```

I'd expect the UHK 80 to not reconnect automatically after disconnecting it, but only upon request.

mondalaci, 2025-01-31:
@benedekkupper Can you reproduce this?

benedekkupper, 2025-02-05:
I can, I have checked this before as well, but couldn't find a solution so far. I made attempts today again, e.g. by keeping only the HID advertisement and removing the NUS data altogether, and also by not advertising while connected to macOS, but both failed to fix the problem. Since there's so much BT logic here from Karel that I feel like walking on eggshells, breaking things I don't understand, my next approach will be trying to create a minimal nordic devkit setup that reproduces this behavior.

benedekkupper, 2025-02-13:
It's a generic problem, [according to Nordic](https://devzone.nordicsemi.com/f/nordic-q-a/118904/ble-peripheral-hid-always-immediately-reconnects-on-macos) this is not something that can be fixed on the device side. We can either raise the issue to Apple, or find a similar product that doesn't reproduce this error, and keep pushing Nordic for a solution.

mhantsch, 2025-02-14:
Would it be of any help to test a Supermini nRF52840 board running ZMK in the same scenario? (My wife has a Macbook.)

mondalaci, 2025-02-14:
@mhantsch Yes, please!

mhantsch, 2025-02-14:
OK, so just to make sure I understand the test scenario:

Pair the keyboard to the MacBook via Bluetooth.
Type a few characters.
On the MacBook, click "disconnect".
Observe whether the keyboard and MacBook reconnect immediately.

Is that the test case?

mondalaci, 2025-02-14:
Yes, but typing a few characters may not even be needed.

mhantsch, 2025-02-23:
OK, tried this with my wife's MacBook and my ZMK-powered ErgoTravel keyboard. It's exactly the same behaviour as you described. After pairing and using the keyboard, when I click "disconnect" in the MacOS Bluetooth settings, it immediately connects again.

I could only stop this by clearing the pairing information on the keyboard (ZMK offers a "clear BT profile" action) or "Forget...." the device on the Mac side.