UltimateHackingKeyboard/firmware #991: Bundle proper config.bin files

mondalaci, 2024-12-12:
Currently, `make-release.mjs` puts the same `config.bin` to every `device/*` directory.

Use the relevant user configurations instead for `uhk60-right`, `uhk60v2-right`, `uhk-80-left`, `uhk-80-right`. I think all these configs should differ now that device id is part of the user config.

Why have we put device id into the user config? I can't remember.

Don't use any user config for `uhk-dongle`.

mondalaci, 2024-12-12:
On second thought, this issue is still relevant. Ideally, per-device configurations should only include the keys of actual devices; otherwise, excess keys take up precious configuration space, so they shouldn't be included.

We should have a master configuration containing every key on every released keyboard. Then, Agent's build process should create per-device configurations, deleting keys not present on devices.

kareltucek, 2024-12-12:
Another option (which I think would be the best in the long run) is to implement a user interface that allows the user to trim down the user config, similar to the way we allow the user to trim the per-key-rgb.

mondalaci, 2024-12-12:
The suggested option is described in https://github.com/UltimateHackingKeyboard/agent/issues/2431

Implementing this issue would be worthwhile so the user configuration is minimal by default.

ert78gb, 2024-12-13:
> On second thought, this issue is still relevant. Ideally, per-device configurations should only include the keys of actual devices; otherwise, excess keys take up precious configuration space, so they shouldn't be included.

> We should have a master configuration containing every key on every released keyboard. Then, Agent's build process should create per-device configurations, deleting keys not present on devices.

This is the current functionality. I think this story can be closed