UltimateHackingKeyboard/firmware #999: Cannot pair UHK80 via Bluetooth to my Linux laptop via settings UI

mhantsch, 2024-12-12:
Framework Laptop 13 running Pop_OS! 22.04 (based on Ubuntu 22.04). 

I am trying to connect & pair the UHK80. It shows up in the Bluetooth settings as "Not set up" but when I click on it, nothing happens. I don't get prompted with pairing code, nor does the UHK show anything.

Pairing works fine from my Android phone.

If I try manually from commandline, I see this:

```
(base) max@max-framework:~/src/lampster$ bluetoothctl pair D2:89:0A:CA:61:05
Attempting to pair with D2:89:0A:CA:61:05
[CHG] Device D2:89:0A:CA:61:05 Connected: yes
[CHG] Device D2:89:0A:CA:61:05 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
```

However, if I manually just issue a "connect", it then starts pairing:

```
(base) max@max-framework:~/src/lampster$ bluetoothctl connect D2:89:0A:CA:61:05
Attempting to connect to D2:89:0A:CA:61:05
[CHG] Device D2:89:0A:CA:61:05 RSSI: -53
[CHG] Device D2:89:0A:CA:61:05 Connected: yes
Connection successful
```

So I finally got it to pair and it works fine now, but I couldn't have done that through the UI.
Seems like there is still room for improvement.

Let me know how I can provide you with further traces, or anything else you need.

mhantsch, 2025-02-01:
For reference, for my ZMK keyboard, compiling the ZMK firmware with `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y` fixed all Bluetooth related issues. It connects, pairs, and works flawlessly and reliably with multiple hosts running Linux and Windows 11.

The UHK 80 still doesn't pair to the same Laptop.

mhantsch, 2025-03-27:
I tested again with firmware 13.0.1. The UHK 80 is connected via USB cable to the Laptop, a Dongle is also connected but not used. The two halves are connected via bridge cable.

When I try to pair the UHK 80 with my Laptop, it reboots as soon as I initiate the connection/pairing from the Laptop. For a brief second, the Pairing code message shows up on the OLED display, but then the UHK reboots. After the reboot, it shows Dongle connection on the display, although it is plugged into USB. After I disconnect and reconnect the USB cable, the OLED shows USB connection again.

Log:

```
(base) max@max-framework:~$ stty -F /dev/ttyUSB0 
speed 9600 baud; line = 0;
-brkint -imaxbel
(base) max@max-framework:~$ stty -F /dev/ttyUSB0 115200
(base) max@max-framework:~$ stty -F /dev/ttyUSB0 
speed 115200 baud; line = 0;
-brkint -imaxbel
(base) max@max-framework:~$ rm log.txt
(base) max@max-framework:~$ cat /dev/ttyUSB0 | tee log.txt 
Bt: connected n/a (n/a, bc:09:1b:f4:dc:87), 0 10
Bt: Bt connected to unknown. Starting discovery.
==== btManager: should scanAndAdvertise 0 1
Bt: Starting advertising, try 0!
Bt: Adv: HID "and NUS" advertising successfully started
Bt: Received passkey pairing inquiry.
Bt: Type `uhk passkey xxxxxx` to pair, or `uhk passkey -1` to reject
W: conn 0x2000bff8: not connected
Bt security failed: n/a (n/a, bc:09:1b:f4:dc:87), level 1, err 9, disconnecting
W: The in-progress pairing has been deleted!
Bt security failed: n/a (n/a, bc:09:1b:f4:dc:87), level 1, err 9, disconnecting
Pairing of auth conn failed because of 9
Pairing failed: n/a (n/a, bc:09:1b:f4:dc:87), reason 9
Bt: Service discovery completed, connection wasn't matched.
Bt: Bt disconnected from n/a (n/a, bc:09:1b:f4:dc:87), reason 35
W: conn 0x2000bff8: not connected
ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/l2cap.c:700
E: r0/a1:  0x00000004  r1/a2:  0x000002bc  r2/a3:  0x20000000
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x0003a2c5
E:  xpsr:  0x01000000
E: r4/v1:  0x00003b35  r5/v2:  0x2000e378  r6/v3:  0xffffffff
E: r7/v4:  0x20012718  r8/v5:  0x20032598  r9/v6:  0x000713e9
E: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x20032548
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x0006ecf2
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x2000e378 (sysworkq)
E: Resetting system
*** Booting nRF Connect SDK v2.8.0-6b415f6ea872 ***
*** Using Zephyr OS v3.7.99-7490142e248f ***
----------
UHK 80 Right USB started
Enabling UART
Uart, unexpected byte: 0
Uart, unexpected byte: 0
Uart, unexpected byte: 0
# (... many of these messages ...)
Uart, unexpected byte: 0
Uart, unexpected byte: 0
UART_RX_DISABLED
uhk8I: 3 Sectors of 4096 bytes
I: alloc wra: 2, df8
I: data wra: 2, c90
I: SoftDevice Controller build revision: 
I: fe 2c f9 6a 7f 36 22 2e |.,.j.6".
I: a0 79 c0 40 be 2c 03 20 |.y.@.,. 
I: 40 c2 f3 32             |@..2    
I: HW Platform: Nordic Semiconductor (0x0002)
I: HW Variant: nRF52x (0x0002)
I: Firmware: Standard Bluetooth controller (0x00) Version 254.63788 Build 573996906
I: No ID address. App must call settings_load()
Settings: Found peer 'left' with address 2fef518fcaed
I: Identity: D2:89:0A:CA:61:05 (random)
I: HCI: version 6.0 (0x0e) revision 0x104e, manufacturer 0x0059
I: LMP: version 6.0 (0x0e) subver 0x104e
Reading hardware config
Reading user config
Applying user config
Flashed User Config version: 9.0.0 (native version: 9.0.0., at v13.0.1 / 5d3ade7cc2af51675007ca4f04aa6000)
Clearing bonds
 - Keeping a known bond.
 - Keeping a known bond.
Resetting left right link! Unidirectional
User config applied
Bt: NUS Server module initialized.
Scan module initialized
Bt: NUS Client module initialized
Bt: Starting bluetooth services.
==== btManager: should scanAndAdvertise 1 1
Bt: Starting advertising and scanning, try 0!
Enabling UART
USB config changing to Hid_NoGamepad
Filters matched: left (n/a, ed:ca:8f:51:ef:2f), connectable:1
Scan connecting: left (n/a, ed:ca:8f:51:ef:2f)
Conn state: UartLeft(Ready)
Resetting left right liBt: connected left (n/a, ed:ca:8f:51:ef:2f), 29 6
nk! Resetting left right link! Unidirectional
Bidirectional
W: conn 0x2000be58 failed to establish. RF noise?
Bt security failed: left (n/a, ed:ca:8f:51:ef:2f), level 1, err 9, disconnecting
Bt: Bt disconnected from left (n/a, ed:ca:8f:51:ef:2f), reason 62
==== btManager: should scanAndAdvertise 0 1
Conn state: Host0(Framework Laptop USB, Ready)
Conn state: Host0(Framework Laptop USB, Ready), Active
Bt: Starting advertising, try 0!
Bt: Adv: HID "and NUS" advertising successfully started
Bt: connected n/a (Dongle UHK80, d5:82:81:0d:e5:96), 2 4
Bt: NUS peripheral connection is ready.
Conn state: Host1(Dongle UHK80, Connected), Peer host1
Bt: connected to host1 (Dongle UHK80, d5:82:81:0d:e5:96)
==== btManager: should scanAndAdvertise 0 1
Bt: Starting advertising, try 0!
Bt: Adv: HID "and NUS" advertising successfully started
Bt: host1 (Dongle UHK80, d5:82:81:0d:e5:96) conn params: interval=7 ms, latency=0, timeout=1000 ms
Conn state: Host1(Dongle UHK80, Ready), Peer host1
Conn state: Host1(Dongle UHK80, Ready), Peer host1, Active
0:right$ Dongle (host1 (Dongle UHK80, d5:82:81:0d:e5:96)) and right half run the same dongle protocol version 1.0.2
```
(display now shows Dongle connection)
```
Conn state: Host0(Framework Laptop USB, Disconnected)
Conn state: Host0(Framework Laptop USB, Ready)
Conn state: Host0(Framework Laptop USB, Ready), Active
```
(display shows USB connection only after unplugging and replugging USB cable)

My preferences in Connections are to use USB:

[image]

mhantsch, 2025-05-16:
Just letting you know that this issue still persists in firmware v14.0.0. This is the log I get when trying to BT-connect to my UHK80 from my Laptop:

```
Bt: start advertising
[00:03:28.852,661] <wrn> bt_conn: conn 0x2000cfa8: not connected
[00:03:28.859,436] <wrn> Bt: Bt security failed: n/a (n/a, bc:09:1b:f4:dc:87), level 1, err 9, disconnecting
[00:03:28.859,466] <wrn> bt_smp: The in-progress pairing has been deleted!
[00:03:28.859,802] <wrn> Bt: Bt security failed: n/a (n/a, bc:09:1b:f4:dc:87), level 1, err 9, disconnecting
[00:03:28.859,832] <wrn> Bt: Pairing of auth conn failed because of 9
[00:03:28.875,732] <wrn> Bt: Pairing failed: n/a (n/a, bc:09:1b:f4:dc:87), reason 9
[00:03:28.883,087] <wrn> bt_conn: conn 0x2000cfa8: not connected
ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/l2cap.c:700
[00:03:28.883,178] <err> os: r0/a1:  0x00000004  r1/a2:  0x000002bc  r2/a3:  0x0000000c
[00:03:28.883,209] <err> os: r3/a4:  0x00000004 r12/ip:  0x00000010 r14/lr:  0x0003d1a9
[00:03:28.883,209] <err> os:  xpsr:  0x21000000
[00:03:28.883,239] <err> os: r4/v1:  0x00003b35  r5/v2:  0x00000000  r6/v3:  0xffffffff
[00:03:28.883,239] <err> os: r7/v4:  0x200330a8  r8/v5:  0x00000000  r9/v6:  0x200330d8
[00:03:28.883,270] <err> os: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x20033080
[00:03:28.883,270] <err> os: EXC_RETURN: 0x0
[00:03:28.883,300] <err> os: Faulting instruction address (r15/pc): 0x00074b92
[00:03:28.883,331] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:03:28.883,361] <err> os: Current thread: 0x2000e700 (sysworkq)
[00:03:29.150,329] <err> fatal_error: Resetting system
uhk80:right$ 
--- 495 messages dropped ---
Uart, unexpected byte: 0
Uart, unexpected byte: 0
Uart, unexpected byte: 0
Uart, unexpected byte: 0
Uart, unexpected byte: 0
UART_RX_DISABLED
[00:00:00.097,900] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.097,930] <inf> fs_nvs: alloc wra: 1, f88
[00:00:00.097,961] <inf> fs_nvs: data wra: 1, 1bc
[00:00:00.099,090] <inf> bt_sdc_hci_driver: SoftDevice Controller build revision: 
                                            fe 2c f9 6a 7f 36 22 2e  a0 79 c0 40 be 2c 03 20 |.,.j.6". .y.@.,. 
                                            40 c2 f3 32                                      |@..2             
[00:00:00.104,492] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.104,522] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.104,583] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 254.63788 Build 573996906
[00:00:00.105,621] <inf> bt_hci_core: No ID address. App must call settings_load()
Settings: Found peer 'left' with address 2fef518fcaed
[00:00:00.117,797] <inf> bt_hci_core: Identity: D2:89:0A:CA:61:05 (random)
[00:00:00.117,828] <inf> bt_hci_core: HCI: version 6.0 (0x0e) revision 0x104e, manufacturer 0x0059
[00:00:00.117,889] <inf> bt_hci_core: LMP: version 6.0 (0x0e) subver 0x104e
Reading hardware config
Reading user config
Applying user config
Flashed User Config version: 9.99.0 (native version: 11.0.0., at v14.0.0 / d63c925ebac51c9f585a8c45c652b0bd)
Setting battery mode to 1
Clearing bonds
 - Keeping a known bond.
 - Keeping a known bond.
Resetting left right link! Unidirectional
User config applied
Scan module initialized
Bt: filling scan filters
Setting battery mode to 1
Enabling UART
Notification: :::
Filters matched: left (n/a, ed:ca:8f:51:ef:2f), connectable:1
Scan connecting: left (n/a, ed:ca:8f:51:ef:2f)
USB config changing to Hid_NoGamepad
Conn state: UartLeft(Ready)
Resetting left right link! Bidirectional
Resetting left right link! Unidirectional
[00:00:00.354,797] <wrn> bt_smp: Unexpected SMP code 0x0b
Notification: CMX
Notification: CMX
Conn state: NusServerLeft(Connected), Peer left
Batteries: 68% 0% (3855 0)
Conn state: NusServerLeft(Ready), Peer left
Bt: start advertising
[00:00:01.394,256] <inf> Battery:   - Resistances 175 232954 107611, brightness 12, correcting 3944 -> 3950
[00:00:01.394,470] <inf> Battery: ... Voltage corrected because of load 3944 (95) -> 3950 (96)
[00:00:01.394,653] <inf> Battery:     ... Averaged to 3950 (96)
Batteries: 68% 96% (3855 3950)
[00:00:02.594,024] <inf> Battery:   - Resistances 175 232954 107611, brightness 12, correcting 3946 -> 3952
[00:00:02.594,116] <inf> Battery: ... Voltage corrected because of load 3946 (95) -> 3952 (96)
[00:00:02.594,207] <inf> Battery:     ... Averaged to 3952 (96)
```

mhantsch, 2025-05-22:
And also still persists in firmware v14.0.1.

kareltucek, 2025-06-06:
I have delved into it today in hopes that it might also shed some light on https://github.com/UltimateHackingKeyboard/firmware/issues/1228 . 

@benedekkupper what are your conclusions and hypotheses so far? 

I have recorded my observations, logs and sniffs at https://devzone.nordicsemi.com/f/nordic-q-a/122025/ble-pairing-with-linux---zero-distribution-flags . TL;DR: there might be a c2usb/GATT problem which interferes with pairing.

benedekkupper, 2025-06-06:
I haven't dug deep into this yet, the only thing I tried was using the same Kconfig options that ZMK was using to circumvent a very similar behavior. That attempt (#1108) didn't yield any results, so I was in the dark on how to continue. I would add that the HID GATT peripheral is not generating any traffic by itself, all of it is either initiated by the GATT central (read or write), or triggered by the user (button presses send GATT notifications).

mhantsch, 2025-08-14:
Should I even try this build, since in #1228 you wrote "Sorry about this build, guys! It clearly has major issues."?

benedekkupper, 2025-09-04:
Here is a new candidate for fixing this issue: https://github.com/UltimateHackingKeyboard/firmware/actions/runs/17463730738/artifacts/3926901498, if you have the chance, please give it a try and provide feedback.

mhantsch, 2025-09-04:
OK, I flashed the latest master (UltimateHackingKeyboard/firmware: 15.1.0 #e3f4c13) from https://github.com/UltimateHackingKeyboard/firmware/actions/runs/17469006015 and now mouse also works over BT.

Finally, we can close this long-standing issue!

closed 2025-09-04 by mhantsch

mhantsch, 2025-09-04:
What's the official process to unpair a device from the UHK? Also, how do I connect another BT devices (for example, my phone) when the UHK is already connected to my laptop. I tried a macro with

```
set bluetooth.enabled true
set bluetooth.alwaysAdvertiseHid true
```

but I don't seem to be able to see the UHK again for pairing on a new device.

I am glad the base connection and pairing seems to be fixed now, but I don't understand the usability. I don't think it's easy to understand how to set up multiple BT connections. It's also weird that when you have a standalone keyboard (no USB connection), you will need to connect USB to be able to use Agent to store the new BT connection and assign it to a profile slot. I know I am tainted by the way ZMK manages this, but still I don't quite understand the UHK logic yet.

mhantsch, 2025-09-04:
Ah, yes, understood. Now my Dongle stopped connecting too, although I didn't change anything to it except unplug it and later replugged it. 🤷

I have to call it a day for now. I'll play it with more another day.