UltimateHackingKeyboard/firmware #1029: Add device command to delete BLE data

mondalaci, 2024-12-15:
It must be possible to delete BLE data from the settings partition.
- Initially, this is useful because of bugs, such as https://github.com/UltimateHackingKeyboard/firmware/pull/822
- In the long term, this is important so that users can remove sensitive information from their devices.

BLE pairing between the halves should be preserved. Everything else should be removed from the settings partition.

@kareltucek This is a blocking issue, so please make it a priority.

mondalaci, 2024-12-17:
I've been thinking about whether this issue is justified.

If the firmware ensures that the settings partition only contains data that is associated to host connections and automatically deletes the rest of the data and the bond between the halves every time upon saves, then this issue has no purpose.

@kareltucek What do you think?

kareltucek, 2024-12-17:
I think it is not necessary, but no trouble implementing it. (Actually, it is already waiting for me to test it before openin g PR.)

mondalaci, 2024-12-17:
If it's redundant, given the firmware's behavior regarding handling the host connection deletions, it makes no sense to have this command.

kareltucek, 2024-12-17:
👍

mondalaci, 2025-01-04:
I'm reopening this issue because it would be useful for troubleshooting purposes.

@kareltucek Please elaborate on the suggested implementation. Maybe the most surefire way would be to wipe the settings partitions and re-pair the halves afterward.

mhantsch, 2025-01-04:
What else is on the settings partition? I mean, this issue literally just asks about clearing the BLE pairing info.

mhantsch, 2025-01-04:
If you clear *everything*, then it's more like "factory reset everything", isn't it?

mondalaci, 2025-01-04:
The settings partition only contains wireless connectivity information, mainly security keys and BLE addresses.

The user configuration is on a different partition, and I'd handle the two separately.

PR #1082 (2025-01-11): Delete all settings
Closes #1029
closed 2025-01-11 by kareltucek