UltimateHackingKeyboard/firmware #1030: UHK80 left half not available if the right half connects first

ert78gb, 2024-12-15:
I am using the https://github.com/UltimateHackingKeyboard/firmware/commit/2734a118934af105a49bab7bb3fa5c69444c8a9f firmware. 

I connect both left and right half to my Macbook via USB cable. Both halves have battery. If I connect first the right half after the left and run the `npm run electron -- -- -- --print-usb-devices` command then only the right half visible to the computer.

I  have to disconnect the right half and reconnect to left half be available to the computer.

The side effect of this behaviour I can't upgrade the firmware with 2 USB cable. Because after firmware upgrade of the right half the left half still not available but Agent shows `To continue the firmware upgrade, now connect the left half via USB. (You can disconnect the right half or use a second USB cable.)` message. But the left half already connected but not visible to the OS.

ert78gb, 2024-12-15:
If I remember correctly it happened earlier with the opposite behaviour, so maybe it is the expected behaviour.

mondalaci, 2024-12-17:
I can't reproduce this issue on my Linux PC. Let's wait for Karel to merge Benedek's large PR, which could affect this issue. Afterward, let's try to reproduce it again.

mondalaci, 2025-02-19:
Nobody has reported this issue, I couldn't originally reproduce it, and the firmware has matured a lot since it's been reported, so I'm closing this issue. We'll reopen it if justified.