UltimateHackingKeyboard/firmware #1032: Issues after firmware upgrade to v12.0.0

jay7x, 2024-12-16:
Today I started UHK Agent and it asked me to update. I allowed it. Then it told me there is firmware upgrade pending (11.2.0 -> 12.0.0). I did the firmware upgrade. Udev rules were outdated, so I updated them also, following the procedure.

Then I realized that touchpad module sends no Left click on tap anymore. I tried to detach and reattach the touchpad module and left keyboard part disappears in the UHK agent. Though, when I re-re-mapped the Left Click on the touchpad module, left part appears again. But touchpad left click doesn't work anyway. `xev` see no click event at all.

I tried to disconnect and reconnect the keyboard with no luck.

Let me know if I miss anything obvious...

Thank you!

jay7x, 2024-12-16:
Left part disappearing is reproducible with every touchpad module "detach, reattach" cycle. Sometimes, Left part appears quickly and disappears immediately after reattaching. Saving changes to keyboard always brings the Left part back.

jay7x, 2024-12-16:
I watched the `journalctl -xef` output and found something I don't think is right. When touchpad is re-attached, I see this as part of the output. See module ID 5 md5 value below:
```json
"rightModuleInfo": {
  "firmwareVersion": "12.0.0",
  "deviceProtocolVersion": "4.11.0",
  "moduleProtocolVersion": "4.3.0",
  "userConfigVersion": "8.3.0",
  "hardwareConfigVersion": "1.0.0",
  "smartMacrosVersion": "3.0.0",
  "firmwareGitRepo": "UltimateHackingKeyboard/firmware",
  "firmwareGitTag": "v12.0.0",
  "firmwareChecksum": "213375ab240c0d38e5259302e3d62263",
  "modules": {
    "1": {
      "md5": "c321d8fb65130ee74d6167d52caf205b"
    },
    "2": {
      "md5": "b9cd4660ab39398e130c14a4077135f7"
    },
    "5": {
      "md5": "  �\u001f\t\u0005"
    }
  }
}
```

jay7x, 2024-12-16:
More debugging info.

Both halves are present in the Agent, detaching touchpad module:
<details>
```
11:34:08.027 › [DeviceOperation] Read "leftHalf" version information
11:34:08.029 › [DeviceOperation] Read "leftHalf" repo information
11:34:08.034 › [DeviceOperation] Read "leftHalf" firmware checksum
11:34:08.035 › [DeviceOperation] Read "leftModule" version information
11:34:08.038 › [DeviceOperation] Read "leftModule" repo information
11:34:08.042 › [DeviceOperation] Read "leftModule" firmware checksum
11:34:08.049 › [DeviceService] Device connection state changed to: {
  "bootloaderActive": false,
  "bleDeviceConnected": false,
  "communicationInterfaceAvailable": true,
  "dongle": {
    "bootloaderActive": false,
    "multiDevice": false,
    "serialNumber": ""
  },
  "leftHalfBootloaderActive": false,
  "hasPermission": true,
  "halvesInfo": {
    "areHalvesMerged": false,
    "isLeftHalfConnected": true,
    "leftModuleSlot": 2,
    "rightModuleSlot": 0
  },
  "hardwareModules": {
    "moduleInfos": [
      {
        "module": {
          "id": 1,
          "name": "Left keyboard half",
          "slotId": 1,
          "i2cAddress": 16,
          "firmwareUpgradeSupported": true
        },
        "info": {
          "moduleProtocolVersion": "4.3.0",
          "firmwareVersion": "12.0.0",
          "firmwareGitRepo": "UltimateHackingKeyboard/firmware",
          "firmwareGitTag": "v12.0.0",
          "firmwareChecksum": "c321d8fb65130ee74d6167d52caf205b"
        }
      },
      {
        "module": {
          "id": 2,
          "name": "Key cluster",
          "configPath": "/add-on/key-cluster",
          "slotId": 2,
          "i2cAddress": 32,
          "firmwareUpgradeSupported": true
        },
        "info": {
          "moduleProtocolVersion": "4.3.0",
          "firmwareVersion": "12.0.0",
          "firmwareGitRepo": "UltimateHackingKeyboard/firmware",
          "firmwareGitTag": "v12.0.0",
          "firmwareChecksum": "b9cd4660ab39398e130c14a4077135f7"
        }
      }
    ],
    "rightModuleInfo": {
      "firmwareVersion": "12.0.0",
      "deviceProtocolVersion": "4.11.0",
      "moduleProtocolVersion": "4.3.0",
      "userConfigVersion": "8.3.0",
      "hardwareConfigVersion": "1.0.0",
[... 344 lines cut ...]
        "vid": 7504,
        "pid": 24865
      },
      {
        "vid": 14248,
        "pid": 2
      }
    ],
    "reportId": 0
  }
}
11:36:34.998 › [DeviceRendererService] dispatch action [device] connection state changed
11:36:35.006 › [DeviceOperation] status buffer segment:
11:36:35.029 › [SmartMacroDocRendererService] downloadDocumentation
11:36:35.030 › [SmartMacroService] start download firmware documentation {
  firmwareGitRepo: 'UltimateHackingKeyboard/firmware',
  firmwareGitTag: 'v12.0.0'
}
11:36:35.031 › [SmartMacroService] firmware documentation downloaded
11:36:35.032 › [SmartMacroService] reference manual downloaded
11:36:35.072 › [DeviceRendererService] dispatch action [device] Status buffer changed
11:36:35.073 › [SmartMacroDocRendererService] dispatch action: [smart-macro-doc] download documentation success
11:36:35.074 › [SmartMacroDocRendererService] dispatch action: [smart-macro-doc] reference manual changed
```
</details>

jay7x, 2024-12-16:
Hmm.. after re-flashing the firmware, I was able to click *once* with the touchpad, after that it stops sending the event again.

kareltucek, 2024-12-16:
@Jopie01, with trackball it is a different issue. 

I can't reproduce it - my trackball's buttons work just fine. Please, provide more detailed instructions, ideally in a new ticket and with a UserConfig ;-).

> You probably are losing everything unless you have backed up your configuration.

Do configs (those from before the upgrade) from your history (in Agent's Configuration tab) not work?

mhantsch, 2024-12-16:
As @kareltucek wrote, Agent automatically stores history:

[image]

Jopie01, 2024-12-16:
This is my configuration ......

[image]

As you can see all the configurations are overwritten. This is after I downgraded back to firmware 11.2 and agent 4.2.1

For the trackball issue I'm opening a new issue.

mhantsch, 2024-12-16:
> I'm just wondering if all the configurations were migrated to the new configuration version

That's not good. That's maybe worth opening another issue for Agent v5.

Jopie01, 2024-12-16:
> Do you mean that the configurations saved before upgrading to Agent 5.0.0 are no longer shown in Agent?

When I upgraded to 5.0.0 I migrated to the new configuration version as well, but I walked into #1035 so I downgraded everything back to the previous state. But because of the new configuration version, I got the error `invalid configuration`. So I had to click on the button `fix configuration` which basically reset the configuration to default values.

They are shown in agent 4.2.1 but are invalid configurations, only the fixed current one it valid.

mondalaci, 2024-12-16:
I see, but the configurations featured in the configuration history have not been overwritten. They are currently invalid until you upgrade to an Agent 5.0.0+ version.

mondalaci, 2024-12-16:
> I haven't dug into it but why are the ALL the configurations migrated to the new configuration version?

They aren't. Every configuration is saved in the actual format. Upon loading them, Agent automatically transforms the configuration to the most recent version. But it can't transform it to earlier versions, hence your issue.

> Is it one big huge blob which contains all the configurations?

These are separate files on the filesystem. See Settings -> "Open configuration folder" in Agent.

We could display which configuration history records are incompatible with the current firmware. But this issue is very rare and isn't worth much effort.

mhantsch, 2024-12-16:
But when reverting back to an older version of Agent, should the user not be able to also load one if those older history configs? I understood from the previous messages that the whole history had been converted, and the older configs could not be loaded anymore even after reverting Agent. ???

mondalaci, 2024-12-16:
This should not be the case. Older configurations should remain accessible. If the saved user configuration version is less or equal than the user configuration version of the firmware, the configuration should be reloadable.

pcooke9, 2024-12-17:
Using **"UltimateHackingKeyboard/firmware: 12.0.0 #c48aa75"**; left-click now works, but assigning the touchpad to anything else (middle/right/back/forward, etc.) only outputs left-clicks. Even assigning "None" action still activates left-click.

Two-finger tap to right-click is also broken. You need to two-finger tap twice, or two-finger tap once, and then single-finger tap once to right-click. Very weird.

This affects both the UHK60 and UHK80.
___

Left half image still disappears from Agent when connecting/disconnecting touchpad module as well. This only happens with the UHK60 connected.

kareltucek, 2024-12-17:
Well, I am confirming that something fishy is going on with history. 

I have just been trying to load my pre-12.0.0 uhk80 configs via history onto the post-12.0.0 uhk80 using agent80 5.0.0, and all my uhk80 configs dissapeared - there are not even the tabs of the devices present.

closed 2024-12-17 by mondalaci

mondalaci, 2024-12-18:
Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12392454444 build resolve your issues, guys?

pcooke9, 2024-12-18:
> Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12392454444 build resolve your issues, guys?

Agent left-half image no longer disappears.

Reassigning TouchPad actions is fixed.

Left-click works normally.

But rwo-finger tap to right-click is still broken for me.

kareltucek, 2024-12-18:
Can you share your user config? Two-finger tap works fine for me with my own config as well as factory settings.

pcooke9, 2024-12-18:
> Can you share your user config?

Just going to bed now, but I reset my config to test it with defaults and it still happens. I'll try to gather more details for you later tonight or tomorrow.

pcooke9, 2024-12-19:
@kareltucek Sorry it took so long, but here's a video showing the TouchPad's two-finger tap to right-click issue:
https://www.youtube.com/watch?v=iYxiAp6P91o

Using an online mouse tester, I finally discovered the two-finger tap's right-click is getting stuck and is being applied to the non-UHK mouse as well. Either tapping the TouchPad normally, or right-clicking the non-UHK mouse unsticks it.

This happens with both the UHK60 and UHK80.

PR #1034 (2024-12-16): Fix touchpad gestures.
Closes #1032
PR #1037 (2024-12-17): Fix connection state issues.
Closes #1032