UltimateHackingKeyboard/firmware #1033: allow configuration of secondary roles with `set keymapAction`

mhantsch, 2024-12-16:
I would like to be able to set the secondary actions from smart macros, not only the primary ones.

I would actually also like to be able to set all kinds of actions, not just macros and keypresses, but that is lower priority.

kareltucek, 2024-12-16:
Why exactly setting secondaries via macros doesn't work for you?

kareltucek, 2024-12-16:
I see.

I guess that using overlayLayer in tandem with replaceLayer doesn't solve the problem?

mhantsch, 2024-12-16:
I would have to create empty layers in Agent to be able to set actions to switch to those layers. If I don't have the layers, Agent doesn't allow me to configure actions for them. I've deleted any layers that I don't specifically have in the keymap. I want to keep it clear. If I replace the layer in `$onKeymapChange`, then I just don't want it in my keymap in Agent. Too confusing otherwise to remember which are the correct layers to configure keys, and which are only "placeholder" layers that will get overloaded.

kareltucek, 2024-12-16:
I see.

mhantsch, 2024-12-16:
I really think it would be very clean and orthogonal if the smartmacro language allowed setting all kinds of keymap actions -- everything that is possible from Agent should also be work through the macro language.